EdgeTX/edgetx #4132: UI - Button inside a group looses focus on ENTER

JimB40, 2023-09-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

as in title

### Expected Behavior

watch video
https://youtu.be/VmNV_3zEHrg

### Steps To Reproduce

watch video
https://youtu.be/VmNV_3zEHrg

### Version

2.9.0

### Transmitter

Radiomaster TX16S / TX16SMK2

pfeerick, 2023-09-29:
tl;dr When using hard key navigation, after changing the state of a button
in a group (button matrix?), the focus goes back to the group/matrix,
rather than staying with the button.

e.g. Model Setup -> Preflight Checks switches, Model Setup -> Inputs ->
edit an input -> more options -> flight modes buttons (the second example
is one of simple toggle on/off buttons, rather than multiple state ones).

Focus should remain on the button after the state change.

philmoz, 2023-09-29:
The default for a button matrix (LVGL control) is to de-focus the entire control after a button is pressed. The same thing happens using touch activation.

If I force focus back on the pressed button inside the button matrix, then you would have to press the RTN key to move focus out of the button matrix every time.

The preflight Switches button matrix is the only one that has multi state buttons - all the rest are either on or off. If we change the behaviour then we are making the UX worse for all the other button matrix cases.

JimB40, 2023-09-29:
[image]
[image]

Here are two more cases with the same problem.
Cant find other button martixes, if there are others point me.

If you try now to set all buttons ON or OFF inside any of above matrixes you will see how daunting and click hungry it is.
Changing  all (or few) button states inside matrix very common 

For Button Matrix - focus should behave differently depending what you use Touch UI or Hardware Keys
1. Touch UI - focus is set to Buttons Matrix after tap
2. Keys - focus is set to selected button inside matrix after ENTER is pressed. To move focus to whole button matrix user has to press RTN.

Anything else (if you have better concept) that reduces number of clicks will do the job.

pfeerick, 2023-09-29:
> If I force focus back on the pressed button inside the button matrix, then you would have to press the RTN key to move focus out of the button matrix every time.

As Robert said, for the hardware keys, this should be the case. i.e. you press ENT to enter the group anyway, so keep in focus until you press RTN to get out of the group.  However I don't see why it being touch vs keys should make it any different... the last touched thing should stay in focus, and that wasn't the matrix, it was the button within the matrix?

PR #4136 (2023-10-16): fix(color): Keep last selected button focused within Button Matrix
Fixes #4132
closed 2023-10-16 by pfeerick