EdgeTX/edgetx #4133: UI - Focused top bar icon gets hiddden when using PAGE> & PAGE<

JimB40, 2023-09-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Focused top bar icon gets hiddden when using PAGE> & PAGE< as navigation method

### Expected Behavior

Fix

### Steps To Reproduce

watch video
https://youtu.be/FaQMvLVUBmw

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

pfeerick, 2023-09-29:
Fix? If you make me watch a video to get the answer or have to get my
crystal ball, I will lose interest in future issues you open. No ain't got
time for that, especially when catching up on issue via email.

*For the active top bar icon to be kept in view when using hard key
navigation*, perhaps?

JimB40, 2023-09-29:
C'mon Peter :) I was using vids 10 years ago workin in QA dept.
Quck & robust way to show what's wrong.
Anyway I will include screen-shots and short description once @raphaelcoeffic helps me on Sunday to coonfigure building env.

pfeerick, 2023-09-29:
That's nice.

Yes, it is quick and robust. As is just a *short* summary. I'm not after a
long essay... just something that can be read before watching the video, or
so can start pre-reading the likely offending code while waiting for the
point to be reached in the video. Unlike Q&A department, we don't get to
charge extra billable minutes. ;) Don't worry about screenshots unless you
want to annotate them... you can just as easily share time-coded links to
the video if you want to highlight a specific screen. Optimise the workflow
you have, rather than creating a new one.

philmoz, 2023-09-30:
Since the 3.0 UI loses all these icons anyway, I don't think there's much point trying to fix this.

pfeerick, 2023-09-30:
For the most part, I agree, particularly as it only happens if you have *all* the tabs enabled... think you only have to disable one tab (which is supposed to be disabled by default anyway) for this to become a non-issue. The only niggle is the question of how far away *is* 3.0, as it could still be a fair way away.

philmoz, 2023-09-30:
Quick solution is to lose the date/time in the menus.

pfeerick, 2023-09-30:
Hm... for model settings at least... radio setup doesn't suffer from this problem... 🤭

pfeerick, 2023-09-30:
You'd be surprised... on weekends don't wear a watch, and phone is usually in the car... so the radio is the timepiece at hand. So it only being on the main screen (either as it is, or via a widget) is perfectly fine by me. But that is only personal preference, and we should probably keep it where it's not causing some other problem. Could just turn the time off if there are more than x tabs enabled (X being the number of tab icons needed to overlap). Thus it becomes visible if there is space, or turned off if in the way.

JimB40, 2023-10-01:
Taskbar with various system information including date. 
- Possibly placed in a bottom part of screen that is difficult to touch
- possibly with global key shortcut to show/hide it.

pfeerick, 2023-10-01:
Sounds like UI 3.0 implementation rather than fixing the immediate problem ;)

philmoz, 2023-10-01:
Another option might be to move one or more of the model pages from the top bar to buttons on the Model Setup page (with the modules, timer, etc buttons).

The 'Custom Scripts' page seems like a good candidate to me and perhaps the 'Heli' page.

PR #5412 (2024-08-13): feat(color): scroll menu bar so selected tab icon is always visible
Scroll the menu bar in the Radio, Model and Screen setup pages to keep the selected tab icon visible.

Fixes issue on NV14/EL18, where selected tab icon goes off screen, when use hat trims as keys and paging left/right to select tabs.

Resolves #4133
closed 2024-08-13 by pfeerick