EdgeTX/edgetx #4134: Splash screen backround  ... must leave

JimB40, 2023-09-29:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

[image]

Guys I know we use what we have but...
This bkg looks: outdated, ugly, incompatible with Eddie ... pick watever you want
It's like to buy new car and put old worn out tyres ...

### Describe the solution you'd like

As there was infinite discussion about new backround ie "I like, I dont like, maybe this maybe that" ...
Just make background PRIMARY3 (dark blue) like on our EdgeTX banner and it will look awesome.

Then disscusion can be continued ...

philmoz, 2023-09-29:
I think black looks better.

[image]

[image]

raphaelcoeffic, 2023-09-29:
@philmoz I'm good with both!

philmoz, 2023-09-29:
Or dark grey. IMO there is too much blue with the blue background.

[image]

raphaelcoeffic, 2023-09-29:
> Or dark grey

Not bad either, it allows for seeing the black shadow.

philmoz, 2023-09-29:
> Not bad either, it allows for seeing the black shadow.

Actually the shadow annoys me a tiny bit as it's not unidirectional - it spreads out in all different directions (not realistic).

JimB40, 2023-09-29:
Hahah I love when it comes to colors.
Whichever you choose it will be better.
IMHO simple black is the best.. preserving battery life ;)

gagarinlg, 2023-09-29:
> > Not bad either, it allows for seeing the black shadow.
> 
> Actually the shadow annoys me a tiny bit as it's not unidirectional - it spreads out in all different directions (not realistic).
> 

There is an invisible light source between you and the logo, so it is realistic.

philmoz, 2023-09-29:
> There is an invisible light source between you and the logo, so it is realistic.

Where can I buy an invisible light source :)

pfeerick, 2023-09-29:
Any particular reason the images are like twice actual size? Just it's making things distorted 👀 

[image]

The grey is nice as it shows the shadowing, but it's not crucial. Power consumption isn't an issue here, but making it black opens up adding something else and making it stand out ;)

Since it seems like we're finally in agreement over #638, how about make the background for both the bootloader and the splash screen black, and add a  box underneath Eddie (push him up if needed) - maybe with a white outline and that blue colour to fill it, which shows the current version (or commit hash if it's not a tagged release)? Which then closes off this issue and #638.

philmoz, 2023-09-29:
> Any particular reason the images are like twice actual size? Just it's making things distorted 👀

Screen capture on retina display (iMac Pro).

philmoz, 2023-09-29:
Applying a bit of color theory - the average color of the logo is a light blue:
[image]

So applied a complementary muted orange as the background (looks better on the radio than the screenshot):
[image]

pfeerick, 2023-09-29:
That I can understand... my only gripe there is that's not one of the colours we use anywhere else 🤣 But I'll leave the final decision on that in Roberts hands.

JimB40, 2023-09-29:
Black

pfeerick, 2023-09-29:
I bumped into this the other day ... `#000000`
```
`#000000`
```

JimB40, 2023-09-29:
.. or blue? 
I really can't decide... feeling blue today ;)

JimB40, 2023-09-29:
[image]
[image]

pfeerick, 2023-09-29:
Black overall seems better, but blue does make the logo stand out a bit
better. Pick one! We could also do a different colour for each major
version? 🤷

JimB40, 2023-09-29:
Black then.
Second later full-colorful UI appears

Black: 
is classy - you can take radio to dinner with your spouse
makes slimer - perfect for a fat boy as TX16S
preserves battery - for 3 seconds only... but we can say later "carbon footprint matters".

last but not least today is Friday and the best Friday is Black

philmoz, 2023-09-29:
Putting the version and codename under the logo requires reducing the size of the logo.
Do you really want to do that?

If so, can you provide resized logo image.

philmoz, 2023-09-29:
There is room to the left and right of the current logo for the version and codename on landscape layout.

philmoz, 2023-09-29:
[image]

pfeerick, 2023-09-29:
That's better. Make use of the unused space on horizontal format. It can be
under on the portrait layout.

philmoz, 2023-09-29:
Do we want to remove the background from the boot loader as well?

pfeerick, 2023-09-30:
Yes please, unless Robert can give a reason it should stay ;) IMO it the background just makes it harder to read.

JimB40, 2023-09-30:
> Putting the version and codename under the logo requires reducing the size of the logo. Do you really want to do that?

Yes .  It's about composition I've been practicing for over 15 yeras as photographer :)

> If so, can you provide resized logo image.

[image]

Here is Splash Logo. 
- size 210x200
- it has black stroke so if we ever use different background color it will have nice separation
- EdgeTX logo is placed to right side so exact position of whole image is:  x=130, y=15

Text lines are CENTERED
- version: font=SMLSIZE, y=218, color=0xFFFFF
- codename: font=NORMAL (capitalized), y=236, color=0x808080 (you may try also SECONDARY1)

JimB40, 2023-09-30:
An if you are able to code this 'magic' it would be very helpful in every-day testing, and providing support

1. Release version - we display codename for end-users (if this is RC we add this info)
[image]
[image]

2. Nighly - we display date (there is only one nighly every day)
[image]

3. PR - we display PR number and build ID, so it's easy to find it in GitHub and be sure we test right version.
[image]

philmoz, 2023-09-30:
I don't like the small logo on portrait screen.
This is a splash screen not a competition photograph entry - I think it is better to keep the large logo and use the space on the sides as Peter said.

JimB40, 2023-09-30:
It not about photography but composition and providing future proof space for data.
Finishing packing myself for the race day as we leave at 6am
If you need anything else GFX-wise I'll be back tomorrow.

philmoz, 2023-09-30:
> * version: font=SMLSIZE

One of the future changes will be to move the fonts to SD storage to free up FLASH space and, hopefully, have all tanslations in one build. 

So only the STD font will be available for the splash screen (also no BOLD font available).

PR #4139 (2023-10-03): feat(color): Remove splash screen and boot loader background, add version info
Also display version and codename on splash screen.

Fixes #4134, resolves #638
closed 2023-10-03 by pfeerick