EdgeTX/edgetx #4135: UI - Prioritize for righty users

JimB40, 2023-09-29:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

As 85% of population is right-handed, as a rule of thumb we place touchable UI elements on the right side and anything that serves as indicators or preview should go on the left 

[image]
[image]

### Describe the solution you'd like

[image]
[image]

### Describe alternatives you've considered

As we have ongoing Input Edit screen #2939 & Curve Edit screen #2953 this can be posponed if we finish them before 2.10 release

Eldenroot, 2023-09-29:
Yes please!

doubleO00, 2023-10-01:
Amen !

sloped-soarer, 2024-07-18:
I suppose if you buy "right handed" radios e.g. Menu wheel on the right hand side  then you might as well completely alienate left handers !.