EdgeTX/edgetx #4149: Radiomaster TX12mk2: USB enumeration fails on 2nd or later connection.

rogerq, 2023-10-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After Power on, on connecting USB to PC and choosing USB appropriate mode, it works correctly. However, on disconnect and re-connect it does not enumerate on PC and doesn't work anymore. A power cycle is required at the Radio Transmitter to get it working on first connect after power ON.

### Expected Behavior

The device should enumerate over USB at every connection and should not require a Power Cycle in between.

### Steps To Reproduce

1. Power ON the Transmitter Radio
2. Plug USB to PC
3. Choose any of the 3 modes. e.g. USB HID
4. Check that it enumerates successfully on Host PC.
5. Unplug and re-plug USB cable to PC
6. Choose any of the 3 modes e.g. USB HID
7. Check that it enumerates successfully on Host PC.

### Version

2.9.0

### Transmitter

RadioMaster TX12 / TX12MK2

### Operating System (OS)

Linux

### OS Version

Ubuntu 22.04.3 LTS

### Anything else?

Host PC dmesg log:
At 1st connect (step 2)
[ 2493.293463] usb 3-13.1: new full-speed USB device number 54 using xhci_hcd
[ 2493.396398] usb 3-13.1: New USB device found, idVendor=1209, idProduct=4f54, bcdDevice= 2.00
[ 2493.396413] usb 3-13.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2493.396419] usb 3-13.1: Product: Radiomaster TX12 Joystick
[ 2493.396423] usb 3-13.1: Manufacturer: EdgeTX
[ 2493.396427] usb 3-13.1: SerialNumber: 00000000001B
[ 2493.401804] input: EdgeTX Radiomaster TX12 Joystick as /devices/pci0000:00/0000:00:14.0/usb3/3-13/3-13.1/3-13.1:1.0/0003:1209:4F54.000A/input/input48
[ 2493.461762] hid-generic 0003:1209:4F54.000A: input,hidraw5: USB HID v1.11 Gamepad [EdgeTX Radiomaster TX12 Joystick] on usb-0000:00:14.0-13.1/input0

At disconnect and re-connect (step 5)
[ 2499.213770] usb 3-13.1: USB disconnect, device number 54
[ 2510.445796] usb 3-13.1: new full-speed USB device number 55 using xhci_hcd
[ 2511.493798] usb 3-13.1: device descriptor read/64, error -32

raphaelcoeffic, 2023-10-02:
I tested this on macOS and it "just works" there. However, on Linux I found that the `hid` does not seem to like the delayed init (due to the popup menu). It works just fine on any number of plug/unplug events when the `USB Mode` is configured directly in the "Radio settings" menu to be `Joystick`. Not sure why it won't complain on the first delayed init though.

rogerq, 2023-10-02:
@raphaelcoeffic , the issue not only happens with HID mode but also with Mass Storage and VCP mode. It has nothing to do with the hid driver on Linux since USB enumeration is failing at GetDescriptor stage, even before HID driver gets a chance to start.

raphaelcoeffic, 2023-10-02:
@rogerq whatever it is, I’m only saying it seems to be caused by late enumeration. That being said, I don’t see it with any chosen mode here anywhere but on Linux.