EdgeTX/edgetx #4153: Add absolute or relative values for input in global functions

Forceu, 2023-10-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have a Jumper T-Lite which does not have a rotary knob and therefore would like to set the speaker volume with a three-position switch with a global function.

Setting volume to maximum is possible with the paramater `MAX`. However setting it to a mid range or low value is not possible. I can only have the volume correspond to this or another switchs position, which results in No sound, somewhat too high volume and max volume.

Even Joshua Bardwell noticed in [this video](https://www.youtube.com/watch?v=CSWXkPCldP8) that it is very hard to simply put a 0 in as value, something slightly above zero is probably not possible at all.

### Describe the solution you'd like

It should be possible to either define values to enter as a parameter for the volume function or have volume/brightness specific values e.g. off/min/medium in addition to the already existing `MAX` value

3djc, 2023-10-02:
You have all the tools already to do that, in at least two different ways (and nany more if you involve GV and such)

A)
Create a 3 point curve that has the 3 values you want.
Create an input based on your switch, name it Vol for example, assign that curve
Create your volume logical switch using NOT the switch directly, but the iVol source

B)
Create an input with 3 lines, name it Vol for example, each having:
- MAX as source
- WEIGHT/OFFSET to whatever you want for each switch position
- in switch field, set switch position for each line
Create your volume logical switch using NOT the switch directly, but the iVol source

Forceu, 2023-10-02:
Thanks a lot for the input, the problem with these solutions however is, that it is all _model specific_, but I would like to create this as a global function. Is there any way to do that globally?

pfeerick, 2023-10-03:
As of 2.9 MIN is also an option. Perhaps for completeness, we should also
have a ZERO / MID constant? Since GFs are by definitin global, we can't
really use anything else, given all the other options are by definition
model specific, and thus not necessarily always available or configured. I
do wonder though why the switches don't seem to be an input option,
allowing for min/max or min/mid/max options. That could just be an
oversight, which I'll try out in a PR I'll be doing soon to fix up the GF
sources...

3djc, 2023-10-03:
Not sure I'm following you @pfeerick . Switches are an acceptable source in GF. Switch DO by definition provide MIN/ZERO/MAX, by what Forceeu seem to want is non linear inputs, where low would be say 30%, mid would be 75% and max 90%

pfeerick, 2023-10-03:
*except when switches ADC PR has blown away switch and pots config, yet again*

3djc, 2023-10-03:
Sorry @pfeerick, I'm missing something, is there something not working I'm not seeing ?

[image]

pfeerick, 2023-10-03:
Simply that when there are no switches configured (due to migrating back and forward between 2.9 and main, and picking just the right PR... which nukes all the switch and pots configs)... no switches are listed! 🤪 Normally that should be the case, which is why I was confused at the time when the radio DIDN'T show them.