EdgeTX/edgetx #4155: UI - GF/SF indicator for Enabled in the list

JimB40, 2023-10-03:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

[image]
[image]

Currently when GF/SF is enabled there is indicator on the list showing it's ON
However it looks like editable/touchable Checkbox UI element which is not.

### Describe the solution you'd like

It can be simple dot as in MODEL/TELEMETRY sensor line.
or square but without frame that suggest it's checkbox.

philmoz, 2023-10-03:
The checkbox (currently only used in the pre-flight checklist) looks like this:
[image]

Nothing like the SF/GF indicator.

JimB40, 2023-10-04:
Quite close visually. Anything that is not looking as UI touchable element will do the job.
What about small icon
[image]

philmoz, 2023-10-04:
I'm not sure a power on/off icon is a good indicator of enabled/disabled state.

JimB40, 2023-10-04:
So may go reversed UI path.
1. By default when creating SF/GF it is 'enabled'. 
- It's bit annyoug you have to enable it while you decided to create one.
- some of GF/SF features have no option to be disabled (always enabled) - ongoing #3024
2. ON/OFF Labal says 'Disabled'
3. This kind of small icon shows when GF/SF is disabled
[image]

philmoz, 2023-10-04:
> So we may go reversed UI scheme as it seems more intuitive.
> 
> 1. By default when creating SF/GF it is 'enabled'.
> 
> * It's bit annyoug you have to enable it while you decided to create one.
> 
> 2. ON/OFF Label says 'Disabled'

So you want to reverse the meaning of the enabled/disabled flag and break every configured SF/GF in existence (that can be disabled).
 
> Users report they want sets of same SF for every model (copy/paste) but to be able to disable them for particular model. This is also true for dev/testing as I want often my screenshot feature to be available only when i'm testing, while I'd like it to be copied with every new model.

Expanding the scope with unrelated issues is a good way to ensure the issue never gets implemented and closed. Especially if you aren't going to bother suggesting solutions. In this case there are at least three ways to achieve the result (copy an existing model, use model templates, or write a lua script to set defaults).

> 3. This kind of small icon shows when GF/SF is disabled
>    [image]

Since not all SF/GF functions can be disabled there should also be an icon for when it is enabled.

philmoz, 2023-10-04:
To get this issue back on topic, you don't like the current enabled/disabled indicator for the SF/GF list display.

Please provide PNG images for an alternative look for both the enabled / disabled states that everyone can agree on.
It would also be worthwhile showing the results in context (modified radio screenshots) so reviewers can get a sense of how it will look in use.
There are three cases to consider - SF/GF active and enabled, SF/GF inactive and enabled, SF/GF disabled.

JimB40, 2023-10-05:
So first things first.
We agreed on: 
- adding enable/disable feature to every function in GF/SF
- that default state of any newly created GF/SF is 'enabled'

1. BW firmware & Companion
In this cases we have one level UI that serves both for GF/SF  status and allows to edit it (sometimes called in-line editing)
 
[image]
[image]

Then checkbox is proper way to inform about if featur is enabled or disabled as anything else would be confusing.
BTW I'd change column label naming from 'Enable' to 'Enabled'

2. COLOR firmware
In this case we have two level UI
- first (LIST) is to inform about status
[image]

- second (EDIT screen) is to change actual values of GF/SF
[image]

### CONCEPT

**GF/SF EDIT screen**
I'm scratching any further discussion about changing labels as this will open Pandora Box :)

[image]

- Label stays as it is (with the same comment that 'Enabled' is better than 'Enable'
- When new GF/SF is created 'Enabled' feature is always ON to avoid situation user doesn't know why 'this is not woring'

**GF/SF LIST**
Here we only have to inform user about status/configuration of GF/SF
'Enabled' is default state. As a rule of thumb if something is default there is no need to display anything (reducing amount of information user recieives)
 
[image]
In above screen
- Every single GF is 'Enabled'
- GF1 is focused
- GF1, GF2 & GF4 are 'Active' that means Functon condition is true

 Now exception is 'Disabled' so we only inform in this case.
 When troubleshooting why GF/SF doesn't work most important is to visually indicate 'Disabled' items
 
[image]

 in above screen
 - GF3, GF5 are disabled - showed two different icons we may use (GF/SF is kind of paused state)
 - of course GF3, GF5 never turns 'Active' (yellow)

philmoz, 2023-10-05:
> * that default state of any newly created GF/SF is 'enabled'

Absolutely not.

SF/GF exists as soon as it is created. 
If it defaults to enabled then it will be processed as soon as you start putting in parameters which could have unintended consequences.

JimB40, 2023-10-05:
Every time SF is created 'Function' is set to 'Ovverride', and when GF is created 'Function is set to 'Trainer' 
I think what you mean is that 'Trigger' is not set '---' so SF/GF is not working 
Still valid to display icon to show this line (GF/SF) is not working.

3djc, 2023-10-05:
What he means is that the function is running as soon as created (created, NOT finalised), and while you select the values, it would actually run all the successive values to your final choice if defaulted enabled when created

philmoz, 2023-10-05:
If you are going to wait for all SF/GF functions to have the enable / disable flag then no icon is needed at all.

Once an SF/GF is disabled it can be shown in the list with the DISABLED text color and border color so it will be consistent with other UI elements.

JimB40, 2023-10-06:
DISABLED was meant and is used (at least should be)  in UI to indicate UI element that not accesible to user (for whatever reason). When DISABLED user can't select this UI element so can't change it.   
This is not the case here. 
Here GF/SF item is accesible, usert can still confifure GF/SF it but it is not 'working" because user decided conciously to pause its executtion. 

WARNING color would be a way to go, but checked with theme like EdgeTX Red and it doesn't work well.
So paused icon to show this GF/SF in not 'enable' as someone coded seems a way to go.

All functions that currently have no 'enable' feature will not display anything.
Once they get this featue they start to display 'pause'.

philmoz, 2023-10-06:
> DISABLED was meant and is used (at least should be) in UI to indicate UI element that not accesible to user (for whatever reason). When DISABLED user can't select this UI element so can't change it. [...]

So leave the border color as the normal color and just change the text color.

> So paused icon to show this GF/SF in not 'enable' as someone coded seems a way to go.

A 'Paused' icon for a disabled state make no sense to me.

philmoz, 2023-10-06:
I think this is easy enough to understand:
[image]

JimB40, 2023-10-07:
Then test it on different themes. particulary the ones with dark background.

philmoz, 2023-10-07:
> Then test it on different themes. particulary the ones with dark background.

It's the theme authors responsibility to ensure their choice of colors works sensibly in EdgeTx.

pfeerick, 2023-10-07:
> It's the theme authors responsibility to ensure their choice of colors works sensibly in EdgeTx.

Yes indeed, it's not up to us to hold their hands in colour choices. 

But sticking to the point of this issue, of the two icons shown, the circled dash makes the most sense, not the pause icon (as it is disabled, not paused, and yes, the language matters, as they are different states and concepts). And I can see the point in only showing that on disabled SF/GF, in order to reduce cognitive workload, as well as not resembling a touchable element. At the same time, I'm quite happy with the current implementation, so am otherwise indifferent. I don't beleive the disabled color should be used in this case, it should only be used when UI elements can't be interacted with.

JimB40, 2024-05-10:
@pfeerick to be closed IMHO. Checkbox is misleading as John stated in review video, but probably it will stay this way.

closed 2024-06-17 by pfeerick