EdgeTX/edgetx #4161: Merge 'Version' and 'About EdgeTX'

philmoz, 2023-10-05:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

It seems redundant to have both Version and About EdgeTX views with overlapping info.

### Describe the solution you'd like

Suggest merging them together, accessed from the About EdgeTX button (remove Version tab from radio settings).

[image]

[image]

Any thoughts?

pfeerick, 2023-10-05:
I have no real preference for one or the other. Other than one being less
detailed, and thus less information overload when you just want to know the
version you are running, or the QR code for the website. You could argue
that with it being on the splash screen that isn't necessary, but that only
applies to a non-custom splash screen (and colorlcd 🤔). One thing that is
conspicuously absent from the color UI which the B&W UI does have is the
URL as a string... for those of use who are *not* glued to our smartphones
;)

The only think I will insist on is consistency... if it's done on colorlcd,
it needs to be done for B&W also.

Message ID: ***@***.***>
>

JimB40, 2023-10-06:
1. We have a little space here so I'd add CODENAME under VERS if build is release or RC (same scheme as in splash)

2. Since QR code is in color. Have you tested it low contrast themes? Does it work well? If not  plain B&W with high contrast ratio will be better.