EdgeTX/edgetx #4170: GPS value does not always update

archmod101, 2023-10-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have a value widget set to GPS telemetry, but it does not update unless I move to another screen and then back to it.

Note: This is on 2.9.1, not 2.9 as highest selectable in Version drop down.

### Expected Behavior

GPS value widget should update as the data comes in.

### Steps To Reproduce

1. Set a value widget to GPS.
2. With incoming telemetry data verified, observe GPS value.
3. Move to another and then back to the screen with the GPS value widget.

### Version

2.9.0

### Transmitter

Radiomaster TX16S / TX16SMK2

pfeerick, 2023-10-07:
Has this changed between 2.9.0 and 2.9.1? As I'm wondering if https://github.com/EdgeTX/edgetx/pull/4071 is the culprit somehow.

archmod101, 2023-10-10:
I don't recall noticing this in 2.9

frankiearzu, 2023-10-14:
I don't think that #4071 is the culprit. The value handling is for specific data types.  Will check out Spektrum.

frankiearzu, 2023-10-26:
Found the problem... by default, anything that is not a single 32bit value will not refresh by default.   We need to do the same trick as in #4071 that is to populate the "value" to something that changes based on the complex datatypes (text, GPS, date_time).  It did not refresh in previous versions either (2.8.5)

While testing this also found that DATE_TIME is not been display correctly in the value widget (color). Fixed it too.
Creating a PU.

PR #4250 (2023-11-04): fix(color): Refreshing of GPS and DATE_TIME telemetry when using Value Widget
Fixes #4170 

Summary of changes:
With data-types who are not plain 32bit numbers, like GPS, Text, DATE_TIME,  we have to populate the telemetry "value" with something that changes so that the "value" widget detects the change.  The trick is to populate the "value" with a hash of the GPS or DATE_TIME complex value. The actual value for GPS and DateTime are stored in different variables. 

The "value" widget was not displaying correctly DATE_TIME, fixed it (don't know why was moving the x coordinate -42 pixels!).
Chan [... cut]
closed 2023-11-04 by pfeerick