EdgeTX/edgetx #4173: Lua model.getInputsCount(input) returns 0 for defined inputs without a source

frank26080115, 2023-10-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

model.getInputsCount(input) returns 0 for an input that has no source specified

### Expected Behavior

model.getInputsCount(input) returns the number of lines defined, no matter if there's a source or not

### Steps To Reproduce

define an input with the companion, do not actually edit anything

run a Lua script with model.getInputsCount(input)

it will return 0 for that index, even though the item shows up in the list of inputs on the radio LCD

### Version

2.9.0

### Transmitter

Radiomaster Zorro

### Operating System (OS)

Windows

frank26080115, 2023-10-07:
I think I have a clue on why this happens

`gui_common.cpp` function `isInputAvailable` uses the macro `if (!EXPO_VALID(expo))`

`api_model.cpp` function `getInputsCountFromFirst` uses `if (!expo->srcRaw || expo->chn!=chn) break;`

perhaps the fix is to use `EXPO_VALID(x)` inside `api_model.cpp` `getInputsCountFromFirst`

just to note `#define EXPO_VALID(ed)          ((ed)->mode)`

`->mode` is set to 3 within function `insertExpo` of `model_inputs.cpp` and also function `InputsPanel::gm_insertExpo` of the companion's `inputs.cpp`. It's never not set, so it is a reliable indicator of if the input exists. In theory, my proposed fix is correct unless it breaks something else.

raphaelcoeffic, 2023-10-11:
> `gui_common.cpp` function `isInputAvailable` uses the macro `if (!EXPO_VALID(expo))`

It seems everyone uses that macro, indeed. So let's use it in `getInputsCountFromFirst()` as well. Feel free to make a PR at your convenience.