EdgeTX/edgetx #4178: Won't save model with " in the name

Kiiski1, 2023-10-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When I duplicate a model and use a " such as 2" quad, the model will be there and then change to something like 'model 8' and the main screen will only have the background image

### Expected Behavior

I expect the model to save properly

### Steps To Reproduce

Duplicate model
Name the new model with a " in it
Select a different model
Go back to the model select menu and the newly created model will be renamed '(Numerical value) model' and will show a blank background image and there will be no data in that model

### Version

2.9.0

### Transmitter

Radiomaster TX16S / TX16SMK2

pfeerick, 2023-10-08:
That should be fixed by #3965, but is waiting for support on Companion side.

closed 2023-11-12 by pfeerick

dizzysizzy, 2023-11-18:
hey, i have the same issue, i want to name my model 5" quad but it just erases all the data on that model after a reboot (tx16s) and im on the newest version of edge tx, is there a way to fix this?

pfeerick, 2023-11-18:
This has been fixed via the referenced PR, and the fix will be part of 2.10.

dizzysizzy, 2023-11-19:
oh, thx, I shall be waiting then :)