EdgeTX/edgetx #4179: rotary encoder on X10s is not accurate

offer-shmuely, 2023-10-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

the rotary encoder does move smoothy on each click

For example, if i try to select model, one click to the right, move one model picture, the second click does not move anything, now the next click may move two model images.
The same happens when trying to select a switch from the combo-box

### Expected Behavior

I expect each “klick” of the rotary encoder to move the selection one element at a time.

### Steps To Reproduce

1. select a page that have many visible buttons
2. move the rotary encoder two click to one direction, and one back
3. notice that sometime it work well, i.e. each click move the focus one button,
4. and sometime the click miss to move the fcous.
5. and sometime it move two button for one click

### Version

2.9.0

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

offer-shmuely, 2023-10-07:
additional info:

- I can see this problem on two X10s tx, one is access version, and one older, non-access
- it does NOT happen on my RM TX16s
- installing edgeTx 2.7.1 make less sensitive, i.e. the problem still appear, but very rare
- installing openTx, the problem disappear

ParkerEde, 2023-10-07:
there may be a connection here with #3445

offer-shmuely, 2023-10-11:
Any thoughts where is the problem?

offer-shmuely, 2023-11-18:
This another sample of the issue,
https://github.com/EdgeTX/edgetx/assets/7026911/dbd09876-da66-4930-8613-0b964e833c29

raphaelcoeffic, 2023-11-19:
I'm sorry but I cannot seem to reproduce any of this my X10, neither on `2.9.x` nor on `main`. I tried a couple pages including "Model Setup", "Outputs" and "Radio Hardware -> Analog" (checking if the counter is moving as expected). I cannot find anything strange or behaving not the way it should.

What I can see on the videos however reminds me of my Microwave which has a heavily battered rotary encoder. As I'm not using the X10 so much these days, I expect the rotary encoder to be in perfect shape.

gagarinlg, 2023-11-19:
Since it is better with OpenTX, what has changed in the rotary encode code since the?

raphaelcoeffic, 2023-11-19:
The base line relevant for X10 is the same.

Here is the OpenTx code:
```c++
  uint8_t newPosition = ROTARY_ENCODER_POSITION();
  if (newPosition != rotencPosition && !(readKeys() & (1 << KEY_ENTER))) {
    if ((rotencPosition & 0x01) ^ ((newPosition & 0x02) >> 1)) {
      rotencValue -= INC_ROT;
    }
    else {
      rotencValue += INC_ROT;
    }
    rotencPosition = newPosition;
```

And here is EdgeTx 2.9:
```c++
  if (pins != state && !(readKeys() & (1 << KEY_ENTER))) {
    if (re_count == 0) {
      // Need at least 2 values to correctly determine initial direction
      re_count = 1;
    } else {
#if defined(ROTARY_ENCODER_INVERTED)
      if (!(state & 0x01) ^ ((pins & 0x02) >> 1)) {
#else
      if ((state & 0x01) ^ ((pins & 0x02) >> 1)) {
#endif
        rotencValue -= INC_ROT;
      } else {
        rotencValue += INC_ROT;
      }

      if (re_count == 1)
      {
        re_count = 2;
        // Assume 1st value is same direction as 2nd value
        rotencValue = rotencValue * 2;
      }
    }
    state = pins;
  }
```

gagarinlg, 2023-11-19:
Any changes regarding pull up or pull down configuration of those pins or the pin speed?

raphaelcoeffic, 2023-11-19:
> Any changes regarding pull up or pull down configuration of those pins or the pin speed?

Not that I know of.

offer-shmuely, 2023-11-19:
I did some more tests,
I took out the sd-card
put a new one, clean, only FIRMWARE folder with list of firmware's
no models, no widgets, no scripts, no images, no voices

install the following versions, one by one:
2.7.1
2.8.0-rc1
2.8.0
2.8.1
2.9.1

results:
2.9.1 - rotary jump/stop from time to time
2.8.0-r1 / 2.8.0 / 2.8.1 - the problem is even more dominant than 2.9  
2.7.1 - have no issue, it work very good

ParkerEde, 2023-11-19:
> 2.9.1 - rotary jump/stop from time to time

it's the same with my x10S express. From time to time the encoder input hangs, then it is normal again and suddenly the marker hangs or jumps. Also during a test under SYS/Hardware of the encoder, the value suddenly jumps to -4 when regularly turning backwards-forwards-backwards (i.e. 0-2-0-2 etc.). Tested with 2.9.2

offer-shmuely, 2023-11-20:
Can you test it with 2.7.1 ?

ParkerEde, 2023-11-21:
I have now tested it with 2.7.1.
I could not find the problem there.

offer-shmuely, 2023-11-25:
I verified that on 2.9.2 (where we have the problem), 
rotary work perfectly in boot mode! (no jump, and no one dead click when changing directions.
just when running the os itself, the rotary is jumping

offer-shmuely, 2024-01-04:
I notice something interesting, 
(Both video taken on the field, one after another, so it is same radio, same version)

When i edit the number on build-in screens, the encoder jump to another area from time to time (very hard to get to specific number i need):

https://github.com/EdgeTX/edgetx/assets/7026911/a90eddf9-49f0-4b8b-ad52-269b8289c6e3

When i edit the number on lua script, the encoder work ok:

https://github.com/EdgeTX/edgetx/assets/7026911/4b8d3490-7f63-4abf-8926-cf9386768f89

So is it still considered problem with my encoder? should i still replace the tx?

offer-shmuely, 2024-09-06:
It became a little better in the last two versions
Not perfect, but usable

closed 2024-09-06 by offer-shmuely