EdgeTX/edgetx #4181: OS wide key emulation for PL18/PL18EV

richardclli, 2023-10-08:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Now we already have this for NV14/EL18:

https://github.com/EdgeTX/edgetx/issues/3202

However, Flysky PL18 is another radio that do not comes with roller and buttons.  Just need a way to map the trims for different radio when the radio need this feature.

Instead of having 2 little joysticks, PL18 has 8 sets of trim switches: TR1 to TR8.  Need some constructs to extend the features of https://github.com/EdgeTX/edgetx/issues/3202 for switches mapping for different radios.

### Describe the solution you'd like

Provide different trim keys mapping:

Based on this PL18 branch:

https://github.com/EdgeTX/edgetx/tree/fs_pl18-fatfs-spi-flash

richardclli, 2023-10-08:
@pfeerick @mha1 @raphaelcoeffic 
I may need some help to start implementing this.

mha1, 2023-10-08:
I think an evaluation is required to determine if a meaningful user interface can be built with scattered trims like the ones on the PL18. Gut feeling: the answer will be no.

If you decide to go ahead I'll support your activity but before writing a single line of code I'd insist on having a UI spec first.

pfeerick, 2023-10-09:
Robert should be able to help with the spec as I'm pretty sure he received a PL18 also.

JimB40, 2023-10-09:
Bit dusty but I will have a look. I think I have ancient FW flashed. Where is the last FW for PL18?

richardclli, 2023-10-10:
@JimB40 already send you firmware via Discord.  It should work without hardware modification, just some features missing.

JimB40, 2023-11-16:
No mater what I do. I can't get internal storage to work on my PL18.
But anyway, to emulate keys using trim buttons.
TR5-TR8 due to a phusical placement are binded with sticks trimming.
That leaves 
TR1-TR2 - upper part
TR3-TR4 - mid part
to use - 8 movements

In color target we have 9 keys, ENTER, EXIT (RTN), NEXT, PREV, PAGE_NEXT, PAGE_PREV, SYS, MDL, TELE

Due to TR placement no assignment is particularly well in terms of ease of use, so I thinh that functional grupping will be solution.
TR1 up - SYS
TR1 down- MDL
TR2 up - ENTER
TR2 down - EXIT
TR3 up - PAGE_PREV
TR3 down - PAGE_NEXT
TR4 up - PREV
TR4 down - NEXT

richardclli, 2023-11-16:
I know the problem.  It is already fixed in latest branch.

pfeerick, 2023-11-16:
While _**holding**_ the transmitter, T1 and T2 are comfortably under my thumbs, making it seem like those should hold the most frequently needed keys? How about the following? Making it so that T3/T4 used to get to the page you want to interact with, and T1/T2 used to interact with it? It's not so important when the radio is flat on the table, but seems more intuitive when holding the radio to be like this. 🤷 

TR1 up - RTN/EXIT
TR1 down- ENTER
TR2 up - PREV
TR2 down - NEXT
TR3 up - SYS
TR3 down - MDL
TR4 up - PAGE_PREV
TR4 down - PAGE_NEXT

JimB40, 2023-11-17:
With those trims it rather guessing. We should code it one way and test.
Anyway it will always be bit like using screwdriver to hammer nail.

richardclli, 2023-11-21:
@mha1 I think the spec can change after some use.  First we need an architrcture that can defines the trims to keys mappings for different targets and not relying on hard coding.
@raphaelcoeffic Need something like the generated json and inc files?