EdgeTX/edgetx #4187: TX16S no hardware button and encoder wheel usage possible after "MPM: Scanning protocols" int. RF module settings

ParkerEde, 2023-10-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After starting the radio and opening the internal RF module of the TX16S, a short spash screen appears showing that it is reading "MPM: Scanning protocols". After that, the interface can no longer be operated with the hardware buttons and the encoder wheel, but only with the touch. Only when you have left the settings of the internal HF module via touch (top left corner) will the hardware buttons and encoder wheel work again.

### Expected Behavior

after closing the splash "MPM: Scanning protocols" should buttons and enc. wheel work.

### Steps To Reproduce

1. power on TX16S
2. MDL
3. touch Internal RF Module
4. waiting for "MPM: Scanning protocols"
5. try to use hardware buttons like RTN or Encoder Wheel

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

Win11

### Anything else?

current main and MPM 1.3.3.33 with actual LUAs from MPM Test Branch

pfeerick, 2023-10-16:
Does the referenced PR fix it for you @ParkerEde? I can't repo this exact issue, but I do get an issue whereby I can only scroll between the top two fields in that screen, until it's exited, which it does fix.

ParkerEde, 2023-10-16:
I will try it soon and then report

ParkerEde, 2023-10-16:
RTN is working now. 
But I can only scroll between the first two fields with the encoder wheel. Here you should also be able to select all fields, I think.

ParkerEde, 2023-10-16:
here a video:
https://drive.google.com/file/d/1m3OZ4BgRZo0qaGYvBqnrZ1Nz_6BhVWgJ/view?usp=sharing

philmoz, 2023-10-16:
> RTN is working now. But I can only scroll between the first two fields with the encoder wheel. Here you should also be able to select all fields, I think.

Odd. On my TX16S I can scroll to all fields on the page after the popup closes.

ParkerEde, 2023-10-16:
I use the current main + this PR. Here a video:
https://drive.google.com/file/d/1kGK2VIRR64tLisNpv6BokjkGFczao6Y1/view?usp=sharing

ParkerEde, 2023-10-16:
ohhh sorry. I have made a mistake..... I'll try again....

ParkerEde, 2023-10-16:
it now works properly - as it should. Thanks
I had used the wrong PR at first

pfeerick, 2023-10-16:
Been there, done that, got the t-shirt 🤣 Much appreciated!

PR #4213 (2023-10-16): fix(color): 'MPM Scanning' popup blocking hardware keys
Fixes #4187 

Delay the 'MPM Scanning' popup until the page has finished building.
closed 2023-10-16 by pfeerick