EdgeTX/edgetx #4188: Automatically uncheck a label when checking a second label results in no models

MattCarothers, 2023-10-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Several of my label groups are mutually exclusive.  E.g. I have an "EDF" group for fixed wing EDF jets, and I have a "Heli" group for helicopters.  The union of those two sets is empty, so if I check both boxes, I get no models.  In order to switch from EDF to Heli, I have to first uncheck EDF and then check Heli.

### Describe the solution you'd like

I'd love to be able to select a label and have the radio automatically uncheck the previous label(s) if the resulting model list would be empty.  If selecting multiple labels results in at least one model being displayed, the behavior should remain the same as it is currently.

### Describe alternatives you've considered

Today I just manually uncheck the first label before checking the new one.

### Additional context

Current behavior: selecting a second label results in no models shown because there are no models with both labels.

[image]

[image]

Desired behavior: selecting the second label should unselect the first label in cases where the combination yields no models.

[image]

[image]

MRC3742, 2023-10-11:
A different approach with the same end result would be to have a separate option to allow "Multiple label selections".
If this new option was not enabled (checked) then selecting a new label category would deselect the current one.

If this makes sense? - Rich

pfeerick, 2023-10-12:
Just FYI, this is mentioned in passing in https://github.com/EdgeTX/edgetx/issues/4122, and I thought it was actually mentioned in an older issue, but can't place it. It could even be in the initial discussion for the labels feature, and it was something to be added later, as the current UI controls wouldn't readily allow for that for the touch UI. 

A workaround for this is to use the page hardware keys, as they move from label to label, rather than multi-selecting labels.

MattCarothers, 2023-10-12:
> A different approach with the same end result would be to have a separate option to allow "Multiple label selections". If this new option was not enabled (checked) then selecting a new label category would deselect the current one.

That was my first thought as well, but I do actually have categories that intersect.  Like some of my warbirds are also EDFs, and it's intuitive to be able to select both at the same time.  I couldn't think of a better way to keep both functionalities than to check the results of the union for an empty set.

pfeerick, 2023-10-12:
Another possibility might be to have it so it's a long press to tick more than one label if we can capture that event? So a normal quick press only activates one label at a time, and if you long press on a second label, it adds that label to the filter.

philmoz, 2023-10-18:
> I'd love to be able to select a label and have the radio automatically uncheck the previous label(s) if the resulting model list would be empty.

I don't see how the logic for this would work and not confuse people.
Sometimes when you select a label there would be multiple labels selected, other times previously selected labels would get unselected. I can see this causing complaints.

Also if you had multiple labels selected and the next one results in no models, how does the code decide which of the previous ones to unselect.

jtaylor2, 2023-10-18:
I think the problem is that most people don't understand (or want) how multiselect works.  They don't realize it is a filter for multiple labels and they don't need or use multiple labels  so it's just an inconvenience to have to deselect a label to select a different one.  They were either used to the way the old categories worked or the way single select works most everywhere (you select a new label it deselects the old one radio button style).  The 5 or 6 people using edgeTX at the 4 clubs to which I belong have labels like Electric, Gas, Glow, Glider, Heli.  They just put models in one "category"  they don't do something like tag Heli-Electric or Heli-Glow they just tag them all Heli.  When used that way multiselect requires and inconvenient extra click to deselect the old label.

  I'm sure there are people that use multiple labels, after all somebody decided to implement it as a multiselect filter instead of radio buttons, but it would be interesting to know the percentages.

Astrohog, 2023-11-16:
I agree that most ppl don't use or require the multiselect feature. I don't but I imagine a small percent do.   A long press could be an option.  Long press on a selected label would produce the label menu while a long press on an unselected label would select it.  Another option would be to use a double tap to select a label while deselecting all others.  This action works well for the "model quick select" feature that I use extensively.   FYI this option is turned on at the bottom of the model setup tab.

jtaylor2, 2023-11-27:
How about a radio option to change between multiselect and single select the way  "Model Quick Select" does for models?  That would make labels work more like the old categories did.

My programming was assembler in the 1960's and K&R C in the 70's so I'm struggling with how to add a radio option (g_eeGeneral.labelSingleSelect) but I've done a proof of concept overloading  "Model quick select" to also do "Label single select".

I'll add a diff for model_select.cpp so that anyone who can compile the code can comment on the concept and if it is worth pursuing.
[model_select.cpp.diff.txt](https://github.com/EdgeTX/edgetx/files/13476457/model_select.cpp.diff.txt)

jtaylor2, 2023-11-29:
[model_select.cpp.diff.txt](https://github.com/EdgeTX/edgetx/files/13495004/model_select.cpp.diff.txt)

PR #4532 (2024-01-26): feat(color): Single select option for model labels plus label AND/OR filtering logic.
Fixes #4188 

This is my take on a better way to handle labels. Feedback welcome.

TODO:

- [ ] Companion support (may need to wait for #4406).

Adds new options to control how the label selection and filtering work. A new 'Manage Models' setup view has been added to the Radio Setup page with the options (Model quick select has also been moved to this new page).

There are three new options:
- 'Label select' - 'Multi select' or 'Single select' (Multi select is the default). If Single select is chosen then onl [... cut]
closed 2024-01-26 by pfeerick