EdgeTX/edgetx #4189: Jumper T20S Customizable Switches

Morbidfpv, 2023-10-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Not really a problem but I did see a github request where a user was talking about the 6 switches on the front/bottom of this radio. They were saying the acted like a 6 position switch, ex: kinda like Radiomaster tx16s 6-postion switch. That user wanted them to act more like individual switches.
With the verision I have that is currently how they act like individual switches that can be set to toggle/2-position switches.
I'd actually like for those 6 switches to be one 6-position switch just like the TX16s. Is that possible?

### Describe the solution you'd like

Have a way to make the 6 customizable switches to be one 6-postion switch. Currently they're indiviual switches as far as i can tell.

Scachi, 2023-10-10:
That one ? https://github.com/EdgeTX/edgetx/issues/3984
By adjusting that numbers (1 or 2 in the screenshot)  you can put them into the same group where a button press will then turn off the other custom button states of the same group.

Morbidfpv, 2023-10-10:
Thanks for the quick reply!
Yes that post was the one I was referring too. I currently have them like
this. How should they be if I want those switches to be a 6-position switch?

Scachi, 2023-10-10:
Set the button type to 2 pos and assign them all the same group number 1 should work

Morbidfpv, 2023-10-10:
I guess I gotta do something else on top of that? When I do that it works in the sense that pressing one switch turns the current one off etc. I’m an FPV pilot, when I’m setting a mode in Betaflight it only moves the range from 1000 to 2000, nothing in between. If switch 1 (bottom right) is on it will be 2000, if I press switch 2-6 it will move to 1000, then only go back to 2000 is I press switch 1 again.

I’m not sure what I should pick for Source in Mixes for this AUX channel I’m making.

Scachi, 2023-10-10:
Sorry, my fault. I didn't understand you question correctly at first.
I hope someone else can help you with a correct solution.
Perhaps you can use special functions to override your channel of choice and set it to a value depending on what switch is active.

Morbidfpv, 2023-10-10:
Ok, thank you. I think if where it says 2 pos, could be changed to 3 pos, 4 pos, 5 pos, 6 pos that would be ideal. I see formal support for the jumper T20S will be in 2.10 edgetx.

pfeerick, 2023-10-11:
Support is already present in the nightlies, and I think at the present time the way to do what you want would be via a mixer line that detects which button is active, and sets the value accordingly. I have seen that config posted somewhere, I just can't remember where of the top of my head.

Morbidfpv, 2023-10-11:
> Support is already present in the nightlies, and I think at the present time the way to do what you want would be via a mixer line that detects which button is active, and sets the value accordingly. I have seen that config posted somewhere, I just can't remember where off the top of my head.

Oh, that's good. Thank you for letting me know.

Morbidfpv, 2023-10-11:
Upgraded the firmware to the nightly but don’t see how I can make all 6 of the customizable switches (SW1-SW6) to act like on 6-position switch like how the 1-6 buttons are on a TX16s. Still kinda acts like a 2 position. Pressing  SW1 is one position and pressing SW2-SW6 is the other position. This changes depending on what I make the Source for the AUX channel in Mixes. If I set it to SW6 for example then SW6 is one position and SW1-5 are the other position. I do see a sources that say CYC1, CYC2 CYC3 which I thought might have something to do with it since I can pick numbers 1-3 for switches SW1-SW6.

pfeerick, 2023-10-11:
CYC1 - CYC3 are the cyclic mixes for heli ;)

You could try something like this - six mixer lines, five of which simply replace the other one depending on the switch that is active. However, I'm sure it was JC who mentioned a much neater way to do this. Or am I wrong? @3djc 

[image]

Morbidfpv, 2023-10-11:
> CYC1 - CYC3 are the cyclic mixes for heli ;)
> 
> You could try something like this - six mixer lines, five of which simply replace the other one depending on the switch that is active. However, I'm sure it was JC who mentioned a much neater way to do this. Or am I wrong? @3djc
> 
> [image]

Dude thank you! I don't know what up with mixes etc. This worked, just had to set the offset for the bottom 3 lines to be from top to bottom -40%, -20%, 0%. Thanks again, for now this looks like it should to the trick.

3djc, 2023-10-11:
They are 6 'distinct' buttons therefore can really not be used like a pot (what older 6pos is). So yes it requires mix lines, but also rethink the way you use and send those (it might, or not, make sense to send all 6 on only one channel)

3djc, 2023-10-11:
Closing the issue since the software already features what's asked. There is an Otx wiki covering this here: https://github.com/opentx/opentx/wiki/Function-switches.

I think there is an edgetx one, but I can't seem to find it again

closed 2023-10-11 by 3djc

pfeerick, 2023-10-11:
No problem. I'll just leave this here then as one possible solution to the "function switches as 6POS" question some people with have for the Jumper T-Pro/T20. One more tip: enable the checkbox at the end of one of the function switches lines, so that one switch in the group must be always on... otherwise if you press on the active button a second time, the output will jump to 0% / middle as all the switches in the group are now off. 

Function Switch settings (default, except for Always On is set):
[image]

Mixer setup: (note that the last line is offset 0, which is the default). 
[image]

Morbidfpv, 2023-10-11:
Thanks again. This totally works for my needs. I use it for team racing to
switch between up to 6 quads.