EdgeTX/edgetx #4201: Images in Radiomaster T16s

Ginnycat, 2023-10-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In the system menu images(PNG)files are visible as normal. In ‘manage models’ only one image is visible, when another model image is added and the tx switched and then switched on again the original image vanishes. No image appears in main screen at all .
When you create a new model using the wizard image is added successfully

### Expected Behavior

Add image in images page and it appears and stays in the ‘manage models’ page and main model screen

### Steps To Reproduce

See error

### Version

2.9.0

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

None

pfeerick, 2023-10-19:
> In the system menu images(PNG)files are visible as normal. In ‘manage models’ only one image is visible, when another model image is added and the tx switched and then switched on again the original image vanishes. 

I'm not sure what is happening here... I can't reproduce this on 2.9.1 w/ TX16S. I can browse and assign PNG and JPG model images via the SD browser, or I can set the model image via the model setup page, and the setting is kept if I switch from one model to another. 

> No image appears in main screen at all .
This sounds like you don't have the "Model Info" widget configured, hence why it isn't visible on the main screen. i.e. Have you gone to Screen Settings -> Main view -> Setup widgets on the model in question, and put the "Model info" widget in a large enough widget space? You should be able to see both the name of the model, and the image.

Ginnycat, 2023-10-19:
Thanks for looking into this issue . I will go through the process again
and see if I doing it correctly. Will let you know
Graham

Ginnycat, 2023-10-20:
Hi Peter
Everything is now working fine. It was my stupidity in not using ‘model
info’ as you said plus the system doesn’t like images with a file name
longer than 9 characters! Thank you very much for your help.
Graham

pfeerick, 2023-10-20:
No worries. Yes, there is a limit on the length of filenames for model images which is why fewer images should be shown when you go into Model Setup to set the model image vs browsing the folder in the SD card browser. Glad it's thats sorted out now though! :) 

I think you still found a bug though, as since `LEN_BITMAP_NAME = 14` characters on color lcd radios, and `10` on widescreen B&W (which includes the `.ext` extension and period), leaving it at 10 characters on TX16. Even though `1234567890.png` is in the `Model Settings => Model Image` list and selectable, it's not saved when I just checked on Simulator for 2.9 (and the trailing `g` seemed to be clipped), but is selectable in Companion and is correctly displayed if I pre-configure a model with that image.

Ginnycat, 2023-10-20:
Glad it was not a total waste of your time,thanks again

PR #4232 (2023-10-31): fix(color): Bitmap name truncation if exactly 14 characters long
Fixes #4201

Fix truncation of model bitmap name on color radios.
Cleanup code.
closed 2023-10-31 by pfeerick