EdgeTX/edgetx #4205: Slow and Delay

kochpet, 2023-10-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I think the Slow and Delay function in Mixers may have a bug. If I add the slow function to the SH switch (momentary swich) nothing happens, the servo immediately and quickly returns to the starting position. Only if I set slow and delay the servo slowly returns to its starting position after releasing the switch

### Expected Behavior

If I set a value for the slow or delay function for up or down, the servo should return to its position either delayed or slowly.

### Steps To Reproduce

1. add the slow ore delay function in mixer to a momentary switch
2. push and release the switch
3. the servo will return withou the slow or dely
4. if you add slow and delay the servo returns slowly in the specified time

### Version

2.9.0

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### Anything else?

[image]

This is the setting for the slow function to work.

philmoz, 2023-10-15:
Remove the SH UP Switch setting.

kochpet, 2023-10-15:
I'm sorry, that was a mistake on my part

closed 2023-10-15 by kochpet