EdgeTX/edgetx #4208: Change rotation of Dial/Rotary switch on Horus x10

bigtwin, 2023-10-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I am just starting with ETX, coming from OTX. It's somehow annoying that the effective rotation of the Dial/Rotary switch is just the opposite from what I intuitively expect and I am familiar with from working with OTX.

### Describe the solution you'd like

It would be nice if effective rotation of the Dial/Rotary switch on a Horus x10 would be configurable in the hardware settings of the radio. Just the choice between clockwise or counterclockwise.
I assume the "reverse" rotation is due to the focus on touch screen operation, which the Horus'  and others do not have.

philmoz, 2023-10-15:
On the Radio Setup page you can change the Rotary Encoder Mode to Inverted.

pfeerick, 2023-10-15:
It's also mentioned in the EdgeTX User Manual, which you may find useful if getting started with ETX

https://edgetx.gitbook.io/edgetx-user-manual/edgetx-user-manual/user-manual-for-color-screen-radios/radio-settings/radio-setup

bigtwin, 2023-10-15:
Oh sorry! I read the manual, but I obviously did not realize that the _Encoder Mode_ is what I was seraching for. 
Thx for the hint!

closed 2023-10-15 by bigtwin

bigtwin, 2023-10-16:
However, it seems like this option is not available in ETX Companion 2.9.1 yet.
Therefor I just updated this here just to have it "searchable" and documented.