EdgeTX/edgetx #4221: Support Radiomaster Pocket

MoonKIE666, 2023-10-18:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

EdgeTX Companion does not have support for the Radiomaster Pocket station

### Describe the solution you'd like

I would like them to support the Radiomaster Pocket, so I can use the companion to change things in the models, such as changing the predefined throttle curves.

3djc, 2023-10-18:
Will happen. All new radio use new 2.10 base code not taken into account in companion yet

amcchord, 2023-10-18:
Is there a beta of companion or somewhere we can test or help support the progress on this?

3djc, 2023-10-18:
There will be 2.10 nightlies supporting it later, no eta yet, but it is in my to do list

Sqwrly, 2023-10-18:
Really looking forward to the nightlies for this. Doing in on the TX is not fun.

amcchord, 2023-10-23:
@3djc I would be happy to make a donation to edgeTX if that helps free up some time to make the nightlies available. I also don't want to create extra pressure.

3djc, 2023-10-23:
It is the next task on my to do list anyway :), but as requires testing, it may take some time to get merged into main.

3djc, 2023-10-29:
Sorry @pfeerick did not link the issue in my PR.

Companion support should be available in tomorrow companion nightly

closed 2023-10-29 by 3djc

pfeerick, 2023-10-29:
No problem, didn't notice that either. 

There aren't nightlies for Companion... instead:

For the latest test version of Companion, from the link below for your preferred operating system, pick the most recent successful build (which will have a green checkmark next to it). On the resulting page, the link to download the installer package is under the Artifacts heading near the bottom.

Companion (main branch): 
- Windows =>  https://github.com/EdgeTX/edgetx/actions/workflows/win_cpn-64.yml?query=branch%3Amain
- Mac => https://github.com/EdgeTX/edgetx/actions/workflows/macosx_cpn.yml?query=branch%3Amain
- Linux => https://github.com/EdgeTX/edgetx/actions/workflows/linux_cpn.yml?query=branch%3Amain

amcchord, 2023-11-18:
Thanks so much for this. Do you know when the simulator functionality will be added?

3djc, 2023-11-19:
Simulators for new radios will be added as part of a significant rework of simu handling in companion so that simu are automatically generated. No ETA for that

FlyMarc2, 2023-12-14:
Is it now possible to use the Pocket with Companion?

3djc, 2023-12-14:
Yes, with companion nighly

pfeerick, 2023-12-14:
> For the latest version testing version of Companion, from the link below for your preferred operating system, pick the most recent successful build (will have a green checkmark next to it). On the resulting page, the link to download the installer package is under the Artifacts heading near the bottom.

Companion (main branch): 
- Windows =>  https://github.com/EdgeTX/edgetx/actions/workflows/win_cpn-64.yml?query=branch%3Amain
- Mac => https://github.com/EdgeTX/edgetx/actions/workflows/macosx_cpn.yml?query=branch%3Amain
- Linux => https://github.com/EdgeTX/edgetx/actions/workflows/linux_cpn.yml?query=branch%3Amain

jims1234, 2023-12-22:
Wow.. I can't wait to try it 👍

amznHendes, 2024-07-12:
Pro-tip: you can view the artifacts as explained above, but there is no downloadable link, you need to sign in.