EdgeTX/edgetx #4225: Lua telemetry script unable to get 'Batt' value

zsjoska, 2023-10-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Unable to retrieve the Radio battery level from LUA telemetry script on Radiomaster Zorro EdgeTX 2.9.1,
getSourceValue('Batt') returns nil

### Expected Behavior

getSourceValue('Batt') should return the charge level of the radio battery, as it does when selected as a value on built in telemetry screen.

### Steps To Reproduce

have a lua script which writes the result of getSourceValue('Batt') to the screen.

### Version

2.9.1

### Transmitter

Radiomaster Zorro

pfeerick, 2023-10-19:
For now, you'll probably need to use getSourceIndex() to get the index for "Batt", and then you can use it with `getSourceValue()`. It _seems_ like `luaFindFieldByName` isn't matching that one. 

```lua
local function run(event)
  lcd.clear()

  local battId = getSourceIndex("Batt")
  local b1 = getSourceValue(battId)
  local b2 = getSourceValue('Batt')

  if b1 then
    print(" b1: ".. b1)
    lcd.drawText(0, 10," Batt: "..b1, 0)
  else
    lcd.drawText(0, 10," Batt: nil", 0)
  end

  if b2 then
    print(" b2: ".. b2)
    lcd.drawText(0, 30," Batt: "..b2, 0)
  else
    lcd.drawText(0, 30," Batt: nil", 0)
  end

  return 0
end

return {run=run}

```

zsjoska, 2023-11-10:
Thank you, the workaround works!