EdgeTX/edgetx #4229: Six Position Switch operation different in Radiomaster TS16S and Companion

boomerwing, 2023-10-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I am seeing this problem in the Radiomaster TS16S, EdgeTX 2.9.1.
When I display the values provided by the 6P switches at the top of the TX I see the following values:
SW1: -100; SW2: =88; SW3: -74; SW4: -60; SW5: -47; SW6: -33
When I display the values provided by the 6P switches in Companion I see the following values:
SW1: -100; SW2: =60; SW3: -20; SW4: 20; SW5: 60; SW6: 100

The transmitter responds to the first set of values, Companion responds to the second set of values.  I therefore cannot use the Complanion setup to load into the TX and expect it to work.

### Expected Behavior

I expect to see the Companion output working for the Transmitter.

### Steps To Reproduce

The error is in the difference between the Companion values and the Transmitter values.

### Version

2.9.1

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

philmoz, 2023-10-22:
Make sure your 6POS switch is properly calibrated on the radio.

Re-run the calibration process - when it asks to center all controls put the 6POS switch to position 3 or 4.
When moving all the controls to the end points go through all the 6POS positions three or four times.

boomerwing, 2023-10-22:
Thank you, philmoz. I never thought of that.

closed 2023-10-22 by boomerwing