EdgeTX/edgetx #4230: Lua output as a Logical Switch input is not permanent in EdgeTX 2.9.1

boomerwing, 2023-10-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I am using an Lua script which I call "sayalt". This script just reads the current altitude value, checks to see if it is close to any 100 ft. step in altitude, and outputs a flag to activate a logical switch. The logical switch uses another Edge Logical Switch and Special Function to make a voice announcement of the current altitude which has just been flagged.
The Lua script works fine when the first Logical switch has the Lua script output "flg" as input. But the problem is, Every time I start the new model I have to go into the Logical switches and add the "flg" as input to the Logical Switch.
In Open TX on a Taranis, once I had defined the Logical Switch it stayed defined.

I am using a Radiomaster TS16S with EdgeTX 2.9.1.

### Expected Behavior

I expect the definition of the Logical Switch to be permanent.  I do not want to fix the definition every time I load the model.

### Steps To Reproduce

Each time I load a new model I have to redefine the Logical Switch.

### Version

2.9.1

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

closed 2023-11-05 by pfeerick

boomerwing, 2023-11-22:
I have installed EdgeTX 2.9.2 which was supposed to have fixed this problem.  When I turn the RadioMaster TX16S OFF then ON again  the  problem is still there.  The output value of the Lua script does not remain as the input of the Logical Switch.  Every time I turn on the TX I must go into the Logical Switch and correct the entry before I fly my model.

philmoz, 2023-11-22:
The fix was added to 2.10; but it doesn't look like it was added to 2.9.

boomerwing, 2023-11-23:
OK.  I thought I saw in the description of 2.9.2 that it was included.
I do have a work-around until 2.10 if that is the decision.  I will watch for the fix.

PR #4231 (2023-11-05): fix(radio): Custom mix script as SOURCE not parsed correctly
Fixes #4230

When a custom mix script is used as a source input the value is stored as 'lua(x,y)' in the YAML file.
The ',' inside the brackets confuses the YAML parser and the value is not read back correctly.

This PR fixes the YAML parser to handle this case.