EdgeTX/edgetx #4235: Play value "TIME" wrong after midnight

Ralf-W, 2023-10-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I defined a Global Function to Play Value "Time" when pressing switch 6P5.
Works as expected 23 hours of the day. But between midnight and 1 o clock in the morning the voice says "1 hour" instead of "zero hour".

### Expected Behavior

It would be nice, if "Play value" would interpret the zero value correctly.

_This is an issue of very low priority if it only affects "Play Time".
But it might affect "play value" of other variables with a value of ZERO._

### Steps To Reproduce

Define a global or special function to "Play value" with the variable "TIme".
Set the time in Setup to any value between 00:00:00 and 00:59:59.
Activate the special or global function. 
The hours-value playing is ONE hour instead of ZERO hour.

### Version

2.9.1

### Transmitter

Radiomaster TX16S / TX16SMK2

philmoz, 2023-10-24:
I can't reproduce this on my radios or in the simulator.

Please check the sound file on your SD card at SOUNDS/en/SYSTEM/0000.wav (if you use a different language change 'en' to your chosen language).

Ralf-W, 2023-10-24:
Thanks for the hint. 

Indeed: Using the "en" language files, I find no problem. 
Even using the german firmware version 2.9.1-lang-custom (e84a3651) together with the "en" sound pack results in the correct output of "zero hours and xx minutes" on my Tx.

Using the main (english) firmware version v2.9.1 "Providence" together with the german "DE" sound pack does show the issue and plays "eine Stunde und xx Minuten" instead of "Null Stunden und xx Minuten". The output uses the file 0101.wav instead of 0000.wav and not the plural "Stunden" but "Stunde".

The same error happens when using the german firmware version 2.9.1-lang-custom (e84a3651) together with the "de" sound pack. 

I did check my SOUNDS/de/SYSTEM/0000.wav. Is plays correctly "Null".

philmoz, 2023-10-25:
Thanks for the info. Looking at the code it plays 0101.wav whenever the hour, minute or second value is 0 or 1.
(playDuration function in tts_de.cpp).

I'll leave it to one of the developers who is a native German speaker to work out the appropriate fix.

pfeerick, 2023-10-25:
@mha1 Could you look at this please?

mha1, 2023-10-25:
k

mha1, 2023-10-25:
@Ralf-W here's 2.9.1 based TX16s firmware (DE language build) if you want to confirm #4242 is working:   
[firmware.zip](https://github.com/EdgeTX/edgetx/files/13168579/firmware.zip)

closed 2023-10-27 by pfeerick

Ralf-W, 2023-10-31:
@mha1 Vielen Dank, klappt prima!

PR #4242 (2023-10-27): fix(tts): DE play value "TIME" wrong for 00:xx:xx hours
Fixes #4235 

Summary of changes:
- fixed use of  singular and plural in TIME tts
- removed "and" between hours and minutes (as is implemented for language EN) to make for shorter and more concise "TIME" speech messages