EdgeTX/edgetx #4238: Logical switch absolute change allows negative values

pfeerick, 2023-10-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The V2 value for `|d| >= x` allows negative values. 

### Expected Behavior

That it be limited to positive values on both B&W and colorlcd since an absolute change can never be negative. Companion limits the input for V2 to positive values already. 

Unless someone can come up with a scenario where it actually can be negative?!

### Steps To Reproduce

L04 is the invalid line.

[image]

### Version

2.9.1

### Transmitter

FrSky X9D+, Radiomaster TX16S / TX16SMK2

### Anything else?

This behaviour appears to be present from OpenTX through to the current version of EdgeTX. 

Given the explanation from OTX University ...

> [|d|>=x](https://open-txu.org/glossary/dx-2/)
> 
> This is used to check if the absolute (irrespective of + or -) Delta (the change since last true) value of a (a selectable source) is greater than or equal to x (a chosen value) as set in V2.  It operates the same as “d>x” without the need to specify a positive or negative signed value.

... and that it is an absolute value change, checking for anything less than 0 should always be true, and thus allowing for a -ve value is pointless and confusing.

Was discovered by a user on the rcvideoreviews discord https://discord.com/channels/761870396085108757/854744834698379274/1166530949480194139

philmoz, 2023-10-25:
This should also apply to the '|x|>y' and '|x|<y' functions.

3djc, 2023-10-25:
Not sure this should be fixed. It is pointless yes, confusing no, it clearly does what the math says and user wrote. If you do it for that, then there should be no reason you should not verify all validity. You can for exemple ask for a value near 0 for a 2pos switch. Valid, but will never happen, and the list is endless.

The policy should be we give you tools. If you want to use a screw driven for a nail instead of a hammer, it's your choice.

pfeerick, 2023-10-25:
I see where you are coming from, but I think for some of the more obviously invalid things... I think we need to sanitise the input where possible. For sure, we can't protect against everything. And, in this instance, if nothing else, the fix Phil has done has made the firmware behaviour consistent with the behaviour of Companion, as that already protects from this invalid input. 

On an unrelated side note... I have used a screwdriver as a hammer... and a hammer on a screw 🤣 They were either the correct tools for those particular jobs, or the only tools at hand... both have been true at one time or another...

PR #4239 (2023-11-12): fix(radio): Don't allow negative values for V2 on LS functions '|x|>y', '|x|<y' and '|dx|>=y'
Fixes #4238

Limit V2 range to positive values, where function compares to |V1|.
closed 2023-11-12 by pfeerick