EdgeTX/edgetx #4240: Boxer build error

kobakirill, 2023-10-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Error when assembling the firmware. Error there is not enough flash memory. Although Boxer, unlike all STM 32F407, has 1MB of flash. There is a processor in it  STM32F407VGT6 with 1MB Flash and 192KB RAM.

### Expected Behavior

Firmware build up to 1MB

### Steps To Reproduce

For example, to build a firmware with a Russian translation.

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster Boxer

### Operating System (OS)

Windows

### Anything else?

[image]

pfeerick, 2023-10-25:
Correct. A new linker script is needed for 1MB B&W due to that radio.

rotorman, 2023-10-25:
I just saw this Issue, related to https://github.com/EdgeTX/edgetx/issues/4243

PR #4244 (2023-10-27): feat: use explicit linker script for 1MB flash B/W radios
Adds explicit linker script for B/W radios with 1MB flash memory (e.g. RadioMaster Boxer). This removes the 512 kB limitation on such radios.

Fixes #4240 and #4243.
closed 2023-10-27 by pfeerick