EdgeTX/edgetx #4243: B/W radios with more than 512kB flash don't use correct linker script

rotorman, 2023-10-25:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Including features for B/W radio firmware with more than 512kB flash, results erranous linker script error about binary exceeding the flash limit, when resulting firmware is over 512kB, but would in reality still fit inside 1MB flash.

### Describe the solution you'd like

B/W radios with more than 512kB flash, e.g. RadioMaster Boxer, should use another linker script than https://github.com/EdgeTX/edgetx/blob/main/radio/src/targets/taranis/stm32f4_flash.ld that limits the flash to 512kB.

pfeerick, 2023-10-26:
Duplicate of #4240

philmoz, 2023-10-29:
Isn't this fixed now?

pfeerick, 2023-10-29:
Yes. Issue wasn't linked to the PR.

closed 2023-10-29 by pfeerick