EdgeTX/edgetx #4253: Value widget: value not greyed out after telemetry lost

mha1, 2023-10-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The displayed value of a Value widget with source telemetry sensor is not be greyed out if telemetry is lost giving the wrong impression of a life telemetry sensor.

Note: if a Value widget is defined in the top bar its value is greyed out if telemetry is lost

### Expected Behavior

Firmware versions up to 2.9 grey out the displayed value if there is no active telemetry. I'd expect 2.9.1 to grey out the displayed values if telemetry is lost.

### Steps To Reproduce

1. go to screen settings, user interface, main view 1 (or any other main view)
2. choose a layout and click on setup widgets
3. click on an widget field click setup widgets
4. choose Value, then choose any telemetry sensor your protocol/receiver supports
5. turn on your receiver and observe the value 
6. turn off your receiver and take note of the telemetry lost message
7. observe the color of the displayed value is not changing wrongly indicating a life telemetry value 

### Version

2.9.1

### Transmitter

Radiomaster TX16S / TX16SMK2

mha1, 2023-10-26:
fixed by #4223

closed 2023-10-26 by mha1

pfeerick, 2023-10-27:
Thanks for the confirm :)

mha1, 2023-10-27:
sorry, found your PR after submitting my issue (after checking there wasn't one existing already)