EdgeTX/edgetx #4254: Delay doesn't work in interaction with Flight Modes

sande005, 2023-10-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When Delay is set on any mixer line, if any Flight Mode is unchecked, Delay won't work, no matter what the currently active Flight Mode actually is.

### Expected Behavior

Delay should work kor any assigned flight mode, even if not assigned for all flight modes.

### Steps To Reproduce

In this example, CH1 responds to a change in the source value two seconds after that value changes, as expected. But CH2, which is the same as CH1 except that it has one or more flight modes excluded, ignores the Delay setting and changes immediately.
Code:
CH1     SF Weight(+100%) Delay(u2:d2)

CH2     SF Weight(+100%) Flight modes(FM0:FM0, FM2:FM2, FM3:FM3, FM4:FM4, FM5:FM5, FM6:FM6, FM7:FM7, FM8:FM8) Delay(u2:d2)

### Version

2.9.1

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Operating System (OS)

Windows

### Anything else?

This is a port over from an OpenTX discussion found starting here:[https://www.rcgroups.com/forums/showpost.php?p=51655957&postcount=10106](url)

Confirmed as an issue with Edge here:[https://www.rcgroups.com/forums/showpost.php?p=51658113&postcount=5860](url)

Thanks to user MiamiMike for "Steps to Reproduce" clipped from his post on RCGroups.

Miami-Mike, 2023-10-26:
Links fixed:

This is a port over from an OpenTX discussion found on RCGroups starting [here](https://www.rcgroups.com/forums/showthread.php?p=51655957&postcount=10106).

Confirmed as an issue with EdgeTX [here](https://www.rcgroups.com/forums/showthread.php?p=51657693&postcount=5859).

raphaelcoeffic, 2023-10-26:
Thx gentlemen for the detailed descriptions and isolation of issue! I’ll look into this ASAP.

raphaelcoeffic, 2023-10-27:
@sande005 @Miami-Mike we will need a lot of tests on the #4256. Not sure how to proceed from here.

@MikeBland may I kindly ask you for a review? It's really hard to see the intent of the original code...

raphaelcoeffic, 2023-10-28:
Please note that the same issue occurs if you have *any* switch selected in the same mix line. And yes, even with the `ON` switch.

PR #4256 (2023-11-07): refactor(mixer): improve delays
This PR removes the distinction between following with respect to how delays are handled:
- a conditional mixer line (using a set of flight modes and/or conditional switch),
- a non-conditional one.

Fixes #4254
closed 2023-11-07 by pfeerick