EdgeTX/edgetx #4262: Liftoff shuts down my USB connected TX16S (ETX2.9.1) within 3 minutes and boots into EM

TTGG4400, 2023-10-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I start liftoff on my laptop - I get it ready to fly freeflight.
I start my Tx, and select my liftoff sim model.
I hook up the laptop USB to the Tx top USB C port, and I select joystick mode.  
I then use the Tx to take off in Liftoff, fly a minute or so.
Then my Tx display goes dark, and liftoff says I've lost connection.  
I toggle power to the Tx and it boots up in emergency mode.  
I toggle Tx power again, and it's working normally. 
Then I try again - and the same thing happens.  4x in a row now.
Haven't tried liftoff in over a year - it used to work though.  
Everything else on my Tx works fine.  
Running ETX2.9.1.

### Expected Behavior

The Tx should not boot into EM mode while running Liftoff.  I don't have problems using the USB C port for normal updates to the Tx, BTW.

### Steps To Reproduce

I hook up the laptop USB to the Tx top USB C port, and I select joystick mode.  
I then use the Tx to take off in Liftoff, fly a minute or so.
Then my Tx display goes dark, and liftoff says I've lost connection.  
I toggle power to the Tx and it boots up in emergency mode.  
I toggle Tx power again, and it's working normally. 

### Version

2.9.1

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

Windows 11

TTGG4400, 2023-10-30:
Hi pfeerick.  The power LED on the radio is off.  But the 6 position rotary ch LED is still on.  And none of the switches have any effect, including the power button - until I remove the USB connection to the laptop.  Then when I try your suggestion, of holding the two horizontal trim tabs inwards while powering on - it comes up in normal bootloader mode (no evidence of emergency mode IOW).   Otherwise, it behaves as previously noted, if I don't hole the two horizontal trim tabs inward while powering on.

pfeerick, 2023-10-30:
Ok, so the really does seem like for some reason it just powered off. The 6POS buttons and leds are controlled by their own MCU, and are probably being powered by the top USB port in this instance, while the main MCU is off. 

I wonder @rotorman... do you think the main MCU could crash, and instead of going into EM, could get hung in DFU mode (thereby turning its own power off since it's no longer holding the power on).

philmoz, 2023-10-30:
I think it could be a crash / restart. I get the same behaviour forcing a crash when the USB joystick is connected.
The radio restarts; but because the USB cable is still plugged in it goes straight to DFU mode.
The main LED is off; but the 6POS switch stays on.

rotorman, 2023-10-30:
If the top USB-C of TX16S is connected when STM32F429BIT6 resets, it indeed goes into DFU, as BOOT0 is pulled (via voltage divider circuit) by top USB-C VBUS (the +5V rail) high.

3djc, 2023-10-30:
Yeah, i don't think EM reboot can happen with top USB connected, but the question remains why does IT EM ?? Could we have a copy of MODELS and RADIO ?

TTGG4400, 2023-10-30:
Here are the zip files of my MODELS and RADIO folders
[RADIO.zip](https://github.com/EdgeTX/edgetx/files/13207236/RADIO.zip)

[MODELS.zip](https://github.com/EdgeTX/edgetx/files/13207233/MODELS.zip)

TTGG4400, 2023-10-31:
I found that Velocidrone does the same thing.

3djc, 2023-10-31:
It is not bidirectional, so any software will do it windows side, even no software I guess

TTGG4400, 2023-10-31:
So, do you guys think it's my Tx hardware?

TTGG4400, 2023-10-31:
Hi Philmoz,  I tried a diff USB port.  I've tried with and without a USB hub.  I tried two diff USB cables besides the original.  Nothing changed with the above trials.  I don't have another PC to try it on though.  However, while using Liftoff, I tried wiggling the USB cable a little (nothing) and a lot - and that (wiggling it a lot) caused a typical USB disconnect, but not an EM.  So in this case, I only had to reconnect the cable and select USB Joystick on the Tx, to continue using Liftoff.  So, I don't think it's a loose cable causing it to go into EM.  Thx for the help, BTW.

philmoz, 2023-11-01:
Thanks. I noticed in your model file you are using some widgets, logical switches and special functions.

Can you try creating a new model with the minimum setup (no widgets, etc), and see if that still has the problem.
Also turn off the trainer port in the model settings.

TTGG4400, 2023-11-01:
Philmoz - thx.  I removed all widgets, turned off the timers, turned off the trainer port, made sure USB Joystick was set to classic, reverted to Edge TX default theme, disabled all preflight checks and disabled all the features - and I was able to run liftoff for some 25 min or so.  Had to go a long time to prove it wouldn't fail.  At least I know it's not my hardware now.  Thanks again.

philmoz, 2023-11-01:
Ok, so looks like a problem in EdgeTX then. 

It would be very helpful (if you have time) to try adding things back one at a time to see if the problem can be narrowed down.

TTGG4400, 2023-11-01:
> Ok, so looks like a problem in EdgeTX then.
> 
> It would be very helpful (if you have time) to try adding things back one at a time to see if the problem can be narrowed down.

Philmoz ... I finished my tests.  It appears that if I insert a widget onto the display, it happens.  What I did ... First I enabled the trainer functionality.  No EM.  Then I tried showall widget on the 1x1 display - i got EM. Then I tried gauge rotary on 1x1 - and got EM.  Then I put timers 1 and 2 on a 2x1 - and got EM. Then I removed the widgets from the display and turned everything else back on - and that worked just fine.  I tried this with two different themes - Edge Tx Default and D. V. A Pink.  These made no diff.  It seems that having a widget on the screen is the culprit.

3djc, 2023-11-01:
What widget was that ?

TTGG4400, 2023-11-01:
3djc, turning off everything, then only putting a value widget on the display with rudder stick as the source didn't EM.  However, as I added items back into the mix, such as the showall widget, it still didn't EM.  Not until I'd added a few of them, did it EM.  All I know at this point is it's more likely to EM with a lot of stuff turned on.  But i can't say for sure which items.  Because I cannot reliably duplicate a failure mode.  For now - I'm tired of siming.  Tomorrow, i can give it another shot.  Thx for working on this.

pfeerick, 2023-11-01:
For the ShowItAll widget... any idea what version you're currently running, and if updating it makes any difference (if you open the main.lua file in the WIDGETS/SHOWALL folder, you should see a 'History' heading with dates and version numbers)? Just to rule out a bug in that widget that has been fixed. 

https://rc-soar.com/opentx/lua/showitall/

If it happens with any widget other than Gauge, Model Info, Outputs, Text, Timer or Value (these are all built into the firmware), the issue could somehow be related to Lua widgets.

TTGG4400, 2023-11-01:
Was running this version of show all ... 
2022-10-17  v0.9.10	Min pane height for full info decreased to 169px (for EdgeTX 2.8)
So I loaded the latest, and ran it with everything else also running - and it EM'd in 20s.
Things don't always EM this quickly - the last test, with the old showall and everything running went just over 9 mins, before EM.  Normally that lasts less than 3 mins.  Makes testing arduous, the inconsistent behaviour.

philmoz, 2023-11-01:
Thanks very much for doing this testing - it really helps.
Could I ask you to test just the showall widget (everything else turned off, logical switches, special and global functions, trainer, etc).

TTGG4400, 2023-11-01:
I just went 21 mins, with the value widget on the display sourced by throttle stick - and no EM.  Everything else was enabled.

TTGG4400, 2023-11-02:
Here's the log from my last test ... 
Wednesday, November 1, 2023 6:57 PM
1.     Started with everything enabled, and D.V.A Pink theme
2.     loaded a new Showall and removed main.luac.
3.     It sitll EM'd, twice - 1st time 10s, 2nd time 1:20min.
4.     switched back to Edge Tx Default theme ...
5.     And ... EM after 4 mins
6.     Switched to value widget with throttle source
7.     And ... no EM after 21 mins

philmoz, 2023-11-02:
I wonder if the mixer scheduler change in #3759 might be a factor.

TTGG4400, 2023-11-02:
Philmoz - you still want me to run showall by itself.  If so - I'll do it tomorrow.  Good night.

philmoz, 2023-11-02:
> Philmoz - you still want me to run showall by itself. If so - I'll do it tomorrow. Good night.

Yes, that would be greatly appreciated.

3djc, 2023-11-02:
I noticed your internal module and trainer settings are not off for your liftoff model, can you ensure they are both off for your tests ?

3djc, 2023-11-02:
> I wonder if the mixer scheduler change in #3759 might be a factor.

I'm wondering the same, but we run at 1000Hz for ELRS too, so it should not be an issue

TTGG4400, 2023-11-02:
Latest test run results - basically running the showall widget with absolutely everything disabled of turned off ...
Thursday, November 2, 2023 10:59 AM
    turned off internal RF per 3djc request
    turned off everything else too ... ADC filter, timers 1, 2 and 3 (was always off); Trainer and I disabled all features
    Added showall widget to display
    Set desktop clock to 20 min as countdown timer to time liftoff flights.
    Started liftoff, plugged in USB to Tx, started flying
    And ... EM occurred after 6 min

TTGG4400, 2023-11-02:
Ran the showall widget on main display with all else turned off or disabled test again - this time it EM'd after 2 min.

TTGG4400, 2023-11-02:
Thursday, November 2, 2023 11:47 AM

    Ran the gauge rotary widget sourced by rudder with everything else turned off or disabled - note that this widget was blinking "Disconnected..." on the display.
    And ... EM after 7 mins.

gagarinlg, 2023-11-02:
Hmm, looks like the joystick code might have an issue. This has been partially redone in a PR that is not yet merged.

TTGG4400, 2023-11-02:
Thursday, November 2, 2023 12:03 PM
    ran the gauge widget sourced by throttle with all else disconnected of turned off
    And ... EM'd after 8:30min

philmoz, 2023-11-02:
> Hmm, looks like the joystick code might have an issue. This has been partially redone in a PR that is not yet merged.

Bu why only when a Lua widget is running and only on some systems? 
I've tried this repeatedly on my TX16S with the showall widget and can't get it to crash.
(I've used RealFlight on Windows and Aerofly on Mac - don't have Liftoff)

gagarinlg, 2023-11-02:
This should not dependent on the software that is running on the host. USB joystick communication should be the same.

Maybe we should try with an identical sd card contents?

pfeerick, 2023-11-02:
Just as a reminder (also for myself as the penny only just dropped)... the OP said he's running 2.9.1... so https://github.com/EdgeTX/edgetx/pull/3759 does **not** apply. 

If he is willing, it would be interesting to backup the RADIO and MODELS folders (i.e. I usually just make a copy on the SD card and add the version number... i.e. RADIO29, MODELS29) and try with a nightly build to see if the issue is still lurking or has been somehow resolved. Otherwise, need a zip of the entire SD card, just in case it some wonderfully weird and wacky combination of specific versions of theme + widgets + audio file + model image + config, etc is needed ... I do have Liftoff so would be closer to being able to repo this more completely, but as Malte said the software really shouldn't matter...

pfeerick, 2023-12-24:
Can you try the latest nightly build (backup your RADIO and MODELS folder first) and see if that makes any difference?

Tactical-Badger, 2024-01-25:
I can confirm this problem is still occurring in v2.9.2

I just upgraded my MK I from a very old version a few days ago, I think the previous version was something like 1.6.x or something like that, Never had a problem when using the controller as a joystick. It now frequently crashes. I tried removing all the widgets, problem still occurs. I tried going into the model configuration screen, and it seemed to take longer to crash(7-8 minutes or so) but did still crash and upon reboot going into EM.

A interesting thing I did notice that may or may not be of interest, I was plugged in to my USB and in joystick mode. I deleted all my widgets in this configuration and the radio crashed. After rebooting from the EM, all my widgets were back like I hadn't deleted them or made any changes.

Tactical-Badger, 2024-01-25:
May have jumped the gun slightly on my last post, just saw 2.9.3 has been posted, upgraded the firmware and have been running the radio for the past 30 minutes or so with zero issues. Looks like something was fixed.

Thanks Devs!!!

pfeerick, 2024-01-25:
AFAIK there was nothing specific in 2.9.3 that should effect this, so you could just be  lucky or it may rear its ugly head again. There were changes to the USB drivers that are only present in the 2.10 nightlies that will hopefilly actually fix this. 

Hence the request earlier to try with a nightly and/or get a copy of the SD card content to see if it was a combination of settings/widget versions etc that was responsible...

Tactical-Badger, 2024-01-27:
Yea the issue is still there. 2.9.3 seemed to be working ok, but I'm still having a issue. I just tried the 2.10 nightly 2c01bee and the problem persists at this point. It seems to be doing it less often, but is still totally random when it happens.  We'll get there i'm sure.