EdgeTX/edgetx #4269: 6 positions switch on EdgeTX Companion simulator not work

pacogalvan, 2023-10-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

The 6 positions switch on EdgeTX Companion simulator not work, allways apears as position 1. I am using the TX16S radio.

### Expected Behavior

The correct value in the simulator corresponding to the 6-position switch.

### Steps To Reproduce

1. Select the any option value on 6 position switch on simulator.
2. The EdgeTX Companion simulator always appears with the value 1.

### Version

2.9.1

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Linux

### OS Version

Linux Mint 21.2 "Victoria"

### Anything else?

No changes on companion, I have download the 2.9.1 directly of web page.
Pardon for my bad English, I am from Spain.

philmoz, 2023-10-30:
Can you upload the .etx file here please.

pacogalvan, 2023-10-31:
Hi Philmoz,

This is the file created with the new button on EdgeTX Companion.
I am using EdgeTX companion to learn how to use the TX16S.
I bought it, but it has not arrived yet.
Pardon for my bad English.

Regards .... Paco.

pacogalvan, 2023-10-31:
Here the etx file renamed as zip because is not possible send it.
[RadioMaster-TX16S.zip](https://github.com/EdgeTX/edgetx/files/13213429/RadioMaster-TX16S.zip)

philmoz, 2023-10-31:
Thanks for the extra info. Attachments don't work via email - you have to upload them on the GitHub web site.

However you have confirmed what I thought was the case - this is a limitation in Companion at the moment.
The default calibration data is not correct for the 6 position switch.
There is a fix pending; but I'm not sure when it will get released.

If you want to manually fix the .etx file you can do this:
- Rename the .etx file to .zip
- Unpack the .zip file
- Edit the RADIO/radio.yml file and find the '6POS' settings in the 'calib' section
- Change 'mid' to 773, 'spanNeg' to 5388 and 'spanPos' to 9758
- Pack the files back into a new zip file and change the extension to .etx
- Load the new .etx file into Companion and the 6 position switch should now work.

philmoz, 2023-10-31:
Here is a fixed version. Rename back to .etx and it should work.

[RadioMaster-TX16S-fixed.zip](https://github.com/EdgeTX/edgetx/files/13213494/RadioMaster-TX16S-fixed.zip)

pacogalvan, 2023-10-31:
Many thanks!!!

PR #4271 (2023-11-10): fix(cpn): Default calibration values for 6POS switch.
When creating a new configuration in Companion the default calibration values do not work for the 6POS switches.

This PR sets better default values (taken from my TX16S) and fixes the values when loading a .etx file if they are set to the old default.

Fixes #4269
closed 2023-11-10 by pfeerick