EdgeTX/edgetx #4272: Unstable telemetry with TX16S + internal ISRM with nightly

Salto42, 2023-11-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

ACCES or ACCST reception with multiple sensors results in unstable telemetry.
Works with EdgeTX 2.9.x

### Expected Behavior

Works as with 2.9.x

### Steps To Reproduce

Use a receiver with external sensors.

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

2.10.0-nightly  (2692b38e)  2023-10-28

ParkerEde, 2023-11-03:
Yes its the same with Horus X10S Express. @raphaelcoeffic is informed about this.
You got permanently "telemetry lost" / "telemetry okay", right?

Salto42, 2023-11-03:
Yes, that's exactly it.

robustini, 2023-11-05:
Same issue here.
Tested with R-XSR rx (ACCESS) in FPORT and X10S Express with the latest ISRM firmware.

robustini, 2023-11-07:
https://github.com/EdgeTX/edgetx/assets/3864712/5e2405aa-b34a-4645-8695-6c528504fd2f

pfeerick, 2023-11-07:
I know it's not news you wanted to hear, but my X9D+2019 and Archer GR8 have been sitting here for over five minutes now without a single flicker, warble or rumble about telemetry. Just suggests it may be a colorlcd specific issue.

ParkerEde, 2023-11-07:
@pfeerick Yes, that is correct. This problem only occurs with color radios. There is also no problem with my Taranis QX7S ACCESS.

robustini, 2023-11-09:
No news about that? :-(

raphaelcoeffic, 2023-11-11:
As it turns out, the issue seems to have been introduced by #4112, and from then on, the mixer scheduler was not working properly (packets only every 50ms), so that the telemetry streams was missing a lot of packets (the module would just not send them).

Fixed with 1f06a3074ddb1be5030024a4c646e971c0dea8c6

closed 2023-11-11 by raphaelcoeffic

robustini, 2023-11-11:
> This problem only occurs with color radios

Yes, only with color radios but only with ISRM.
Tested everything today with MPM module and no problem, same rx (but in ACCST) and same tx, same hardware.

raphaelcoeffic, 2023-11-11:
> only with color radios but only with ISRM

Just to set the record straight: only `horus` targets (**not NV14 / EL18**), and **all** modules. We had no working **mixer scheduler**, meaning that control packets were sent only every 50ms. That being said, it seems as well that it does not bother the XJT modules, which will still happily send telemetry at the same interval, thus not causing any "telemetry lost" messages.