EdgeTX/edgetx #4283: Graupner GR-18 V2.04 Failsafe not being set

ZoofPL, 2023-11-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Setting failsafe through the radio works only once after binding. 
Subsequent changes will not be updated to the GR-18 firmware V2.04. 
V2.04 is the latest Firmware provided by Graupner. 

Changing failsafe can be done by a workaround: 
Bind the same receiver to a new receiver number and make changes to the failsafe setting. 
This works only once. Subsequent changes require binding to another number again.

Other receiver models that I tested with the latest firmware; GR-16 and GR-12L have this fixed while they had the same issue with older versions of the firmware. 

### Expected Behavior

Changes to failsafe will automatically be saved to the receiver. 

### Steps To Reproduce

Steps to reproduce: 
1) bind a HOTT GR-18 receiver to a unique receiver nr. 
2) Set Failsafe Mode to "Custom" and press "Set" to set up channels as desired.
3) Wait 10 seconds and test if Failsafe settings were succesfull - they should be stored in the RX as set. 
4) Make some changes to the Custom Failsafe settings.
5) Wait 10 second and test again, the new fail safe settings are not stored in the RX. The channels will show values set at step 3).  

### Version

2.9.1

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

Windows 11 home

### Anything else?

Perhaps this is not a EdgeTX firmware issue but a Multi-Module firmware issue? To be sure same issue reported there.

mha1, 2023-11-05:
This is not an EdgeTX problem because EdgeTX doesn't know which receiver the MPM is bound to.

I also saw this problem at one point in time and think it can occur on receivers previously bound to Graupner radios. To solve it I took my Graupner radio and used it to set failsafe with POS on all channels and 1s delay. After that I bound the receiver again to EdgeTX/MPM. After that setting failsafe worked multiple times without re-binding.

I believe the gyro receivers behave slightly different and need to be reset to factory defaults if the were previously bound to Graupner receivers. Please try to flash 2.04 again as this triggers an automatic factory reset. Then bind it to your EdgeTX/MPM radio and set failsafe to different positions to check if flashing resolved the problem. If that doesn't help try what I described above using a Graupner radio.

Please close this issue as EdgeTX has nothing to do with it and can't solve the problem.

3djc, 2023-11-05:
You should report it to multiprotocol github,since is likely is down the rf protocol

closed 2023-11-05 by 3djc

mha1, 2023-11-05:
Already done

ZoofPL, 2023-11-05:
Thanks for the quick response. To complete the topic: 
Updating the multi-module firmware from aetr-v1.3.3.7 to taer-1.3.3.33 resolved the issue. 

Should have done before but learned about the MM firmware upgrade possibility on the go. 
Again thanks for the great work!

mha1, 2023-11-05:
Oh, yes, I forgot. I made changes to the MPM in 1.3.3.20 for two HoTT protocol problems, one being the "set failsafe only possible once after bind": https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/pull/695

Please close your MPM issue too with a not that updating from 1.3.3.7 to 1.3.3.33 fixed the problem