EdgeTX/edgetx #4295: fr translation

Pat6874, 2023-11-07:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

better translation Fr to fill fields in color screen

### Describe the solution you'd like

#define TR_SELECT_MODEL                "Sélection Modèle"
#define TR_TTL_WARNING                 "Attention: utiliser valeur logique 3.3 V"
#define TR_BATT_CALIB                  "Calib. batterie"
#define TR_ENABLED_FEATURES       TR("Fonctions activées", "Fonctionnalités")
#define TR_STICK_NAMES                 "Dir", "Prof", "Gaz", "Ail"
#define TR_USE_THEME_COLOR              "Utiliser couleur du thème"
#define TR_STACK_MIX                   "Mixeurs: "

raphaelcoeffic, 2023-11-08:
@Pat6874 please make a PR for this, issues are not really meant for proposed code changes.

pfeerick, 2023-11-10:
I'll do a PR for this shortly for you Patrick

Pat6874, 2023-11-10:
Thanks...
Patrick

pfeerick, 2023-12-24:
- `TR_STICK_NAMES`  these are limited to three characters, so no.
- `TR_USE_THEME_COLOR` appears to already have been done. 
- `TR_BATT_CALIB` I had to keep the existing string for smaller screens as they are limited to 11 characters where that is shown.

Sorry for not sorting this out sooner.

closed 2023-12-24 by pfeerick

Pat6874, 2023-12-24:
> via fbe06b3

Hello Peter,
Thanks for your reply.

OK for the limitation of 3 caracters. I suggest:
#define TR_STICK_NAMES "Dir", "Prf", "Gaz", "Ail"
Merry Christmas
Patrick

commit fbe06b36 (2023-12-24): chore(translation): FR updates
Closes #4295