EdgeTX/edgetx #4297: Edge TX 2.9.1 Communication with FRSKY RB30+

Jim7695, 2023-11-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I am using a Radiomaste4r TX16S Mark 2 Max running edge tx version 2.9.1.  Connected to a FRSKY RX8R Pro Receiver, SBus out and Smart Port out to a FRSKY RB30+ redundant bus module.  Receiver is bound to the radio and control has been confirmed, telemetry has been established.  While executing the RB30-40 Parameter Set LUA Script, the information from the module is not being received.  I rolled back my Edge TX firmware to 2.8.5 and ran the Parameter Set LUA Script and all expected data was received.  

### Expected Behavior

Expected data from the RB30 to the radio via the LUA script was not received.  

### Steps To Reproduce

Set radio to Edge TX 2.9.1 firmware, connect to the RB30+ via a receiver SBus and Sport, run parameter set Lua script.  

### Version

2.9.1

### Transmitter

RadioMaster TX16S / TX16SMK2

3djc, 2023-11-10:
Could you try a 2.9 later than oct10 (@pfeerick , could you guide him to get a recent 2.9 ?)

Jim7695, 2023-11-10:
Below is the link to the page within FRSky where I got the scripts.  

https://www.frsky-rc.com/rb-30-plus/

Cheers. Jim

Jim7695, 2023-11-10:
Hi All - I have worked out a resolve to this issue.  Downloaded and flashed a fresh copy of the Edge Tx 2.9.1 firmware.  Downloaded and installed the RB 30-40 scripts from the FrSky web site.  Make sure you have version 2.0 of the scripts.  Started the radio and selected the model I am setting up.  During this test, I did not establish telemetry prior to running the Parameter Set Script.  The script ran and all expected data was received and I was able to change the parameter values.  Next I established the telemetry, using discover new.  all telemetry values expected came to the radio.  Went back to the system settings and ran the Parameter set script again, script runs, no data received.  Went back to the model, telemetry, delete all.  Ran the script again, and all data appeared as expected.  

Conclusion - Parameter set script needs to be configured prior to establishing telemetry.  If you need to make changes to these settings, you will need to delete all telemetry prior to running the script to make the changes.   

Cheers, Jim

pfeerick, 2023-11-11:
Interesting. Will have to have a look at the source of the script to see if can figure out why that is happening. Glad to hear you found a workaround though.

I suggested checking the version as https://www.frsky-rc.com/lua-script/ seems to be offering only v1.01 from the date code, but the link you gave gets you to that later v2.0 which has RB30+ support.

Jim7695, 2023-11-17:
Thanks for the reply.  I stumbled across the work around, but glad I did.  I had to hunt thru the frsky-rc web site to find the newest version of the firmware for the RB30+.  

Cheers. 

Jim (Torch) Anderson