EdgeTX/edgetx #4299: heanding

kasatka60, 2023-11-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Jumpter T12 Pro, Edge Tx 2.9.1, happymodel elrs 868mgz (https://github.com/ExpressLRS/ExpressLRS/pull/2305), inav 6.1.1, X-Plane 11.
why heanding less <(2^15/100)?
[image]

### Expected Behavior

//

### Steps To Reproduce

//

### Version

2.9.1

### Transmitter

Jumper T12

### Operating System (OS)

Windows

ikalnytskyi, 2026-04-26:
I hit the same bug. It seems it's coming from the fact that EdgeTX treats the two-byte GPS heading field as a signed value instead of unsigned. Values above 327.67° (the field is degrees × 100, so anything above 32767) will have the most significant bit set, causing EdgeTX to interpret them as a negative number.

mha1, 2026-04-27:
see PR #7066