EdgeTX/edgetx #4301: movement of servo not lineal. ( little jumps )

tonims1, 2023-11-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

in the last nightly. I haven't tried others. The servo movement is not linear. It works in jumps. If I go back to 2.9 it works fine.

### Expected Behavior

linear operation, without jumps

### Steps To Reproduce

It is detected immediately. any stick movement

### Version

2.9.1

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

windows 11

pfeerick, 2023-11-11:
With what RF link? If you look at the channel monitor, are those same steps reproduced there? If so, does a stick calibration fix it?

tonims1, 2023-11-11:
same plane, just changing the firmware version. They are micro jumps.

https://github.com/EdgeTX/edgetx/assets/7953406/03a6bd26-410d-4c38-9806-812c35b4985f

https://github.com/EdgeTX/edgetx/assets/7953406/a1484b75-9449-4ef6-922f-72b33de3ccdb

3djc, 2023-11-11:
Please retest tomorrow after todays fix

tonims1, 2023-11-12:
it's OK now. Thank you

closed 2023-11-12 by tonims1

3djc, 2023-11-12:
Thanks for your test and feedbacks !