EdgeTX/edgetx #4305: Companion 2.10 - Index column is blank for B&W radios.

philmoz, 2023-11-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

The Index column is blank for B&W radios:
[image]

### Expected Behavior

Should show the model index number.

### Steps To Reproduce

Load a B&W .etx file into Companion 2.10.

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky Q X7 / Q X7S (ACCST)

closed 2023-11-12 by pfeerick

ChrisQuadFPV, 2024-06-20:
My QX7 Access has 2.9.4 When I update to 2.10.0 the bootloader takes but shows "no SD card"

pfeerick, 2024-06-20:
> My QX7 Access has 2.9.4 When I update to 2.10.0 the bootloader takes but shows "no SD card" 

Please create a new issue, or search first, because this does not relate to this issue. If you search for sd card, you should find a couple of issues, whereby some slow to initialise off-brand cards which used to work are not working with 2.10.

PR #4307 (2023-11-12): fix(cpn): Model index missing in model list for B&W radios
Fixes #4305

Fix format string so index value is correctly displayed.