EdgeTX/edgetx #4311: Only 64 Mixer-Lines are possible

AlphaCharlyX, 2023-11-12:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Yes, for a big Sailplane with V-Tail and 6 Flaps 64 Mixer-Lines are not enough.

### Describe the solution you'd like

128 Mixer-Lines. Ethos has 120 Mixer-Lines.

### Describe alternatives you've considered

I use Ethos for that plane.

AlphaCharlyX, 2023-11-14:
Is it a big problem to increase the possible amount of mixer-lines in EdgeTX?

gagarinlg, 2023-11-14:
Did you take a look at the templates made by @jfrickmann ?
I really don't understand why you need more than 8 mixer lines per channel.
On B&W radios we can not  increase the number of mixers, because we are low on RAM. On the color LCD radios it might be possible, but nobody tested it.

AlphaCharlyX, 2023-11-14:
Thank you very much for the answer. I will take a look at these templetes and will give a response then soon.

raphaelcoeffic, 2023-11-14:
> Is it a big problem to increase the possible amount of mixer-lines in EdgeTX?

At the moment, we have a fixed amount of lines (used or not) in RAM. To allow for more without draining some radios' RAM, we need to move to dynamic memory allocation for the mixer lines, and make the YAML parser compatible to this. It *is* on our internal roadmap (at least mine), but it is really not yet doable short-term.

jfrickmann, 2023-11-14:
My first F5J setup needed 65 mixer lines (so I cut a less important one) with a V-tail and 63 without V-tail. As I improved the setup, the number of lines decreased. My SoarOTX template has 53 lines (as far as I just counted) and some are for a special configuration mode, so maybe upper 40s. 

You can find the SoarOTX F5J template over here: https://github.com/jfrickmann/SoarOTX It's for OpenTX 2.3, but Frankie Arzu has just started porting it to EdgeTX, please see here: https://www.rcgroups.com/forums/showthread.php?p=51717543&postcount=464

Good luck!

Helipro, 2023-11-16:
Global Variable and Curves helped for cutting down my mixes when they became to many,  You can also move some mixes to inputs or do mixing in the flight controller.