EdgeTX/edgetx #4322: Possibility to invert the left & right sliders direction

kochpet, 2023-11-14:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

It would be nice if you had the option to reverse the direction of the left and right sliders because on the TX16S they work in exactly the opposite direction as on the X12S, for example.

### Describe the solution you'd like

An option to reverse the direction of the sliders.

ParkerEde, 2023-11-14:
Yeah. This would be nice

pfeerick, 2023-11-14:
And this will be possible in 2.10 (already possible in the nightlies)...
There is a toggle swirch at the end of the row for each pot on the radio
settings -> hardware -> pots configuration screen which allows for that.

closed 2023-11-16 by raphaelcoeffic

psfales, 2024-09-18:
This works for S1 and S2, but apparently does not work for the 6-position switch on the Radiomaster TX16S.   Is that a bug?

I was trying to reverse the 6-position switch so that the power-on default at the left would be the max value instead of the min value.  But when I activate the toggle at the end of the row, the 6-position switch stops working. (It remains locked at the max value no matter which button is selected)

pfeerick, 2024-09-18:
It cannot generally can't work (properly) for the 6POS on the TX16S, for  the simple reason that the power on default will **always** be the first/min value. That is, unless you reprogram the IC chip that is responsible for controlling the 6POS leds, reading the buttons, and simulating an analog input. As to why it locks on max value... I'm not sure... it might need recalibration after doing that, but even so, the LED will never reflect the selected value any more if you do get the inversion to work.

philmoz, 2024-09-18:
Inversion of 6POS has been removed in 2.11.
[image]

psfales, 2024-09-19:
Thanks for the info.   I with it could be possible to invert the 6-pos switch, but if that's not possible, then making that clear by removing the button sounds like the right thing to do!

rotorman, 2024-09-19:
It is possible, just not trivial. See: https://www.rcgroups.com/forums/showthread.php?3855505-Blog-13-RadioMaster-TX16S-6-Position-Switch-firmware and https://github.com/rotorman/TX16S-6Pos/blob/main/main.c