EdgeTX/edgetx #4324: No power when selecting Trainer CPPM in 2.9.2

sande005, 2023-11-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Selecting Trainer Master/CPPM Module does not supply power to the module in use as implemented in #4167

### Expected Behavior

Although tested module uses SBUS and not CPPM, one would expect the trainer option to supply power, even if SBUS is not currently supported.
Selecting any External Module will power the module, although trainer options then limited to Master/Jack or Slave/Jack.

### Steps To Reproduce

Select None for external module.
Select Master/CPPM in Trainer section. 
Restart radio, and note the module bind led's do not light, as they do when external module is enabled.

### Version

Other (Please specify below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Operating System (OS)

Windows

### OS Version

11

### Anything else?

Tested with above TX running version 2.9.2 and a Frsky FWTM wireless trainer module.
Used x10-access-v2.9.2.bin to update TX.

Groove70, 2023-11-14:
I have supply power for the module in MASTER/CPPM, but not for MASTER/SBUS. Transmitter QX7 ACCST  2.9.2

raphaelcoeffic, 2023-11-14:
@sande005 is that possible that it only works when you set up it, but not after a restart?

sande005, 2023-11-15:
So odd...it honestly was not working over the last two days.  Tried all the combinations many times on different days, with restarts everytime I changed something.  No new models, no reflashing.
But, just now, before responding to this thread, gave it one last try (trainer/cppm selected, external module off) ...and the embedded RX is getting power and binding to the slave TX.  Of course no control, but to be expected with needing sbus instead of CPPM.  It continues to work, after 10 power cycles.
Module not dismounted/remounted during all this, but perhaps I have to look closer at pin and socket contacts.

Just for curiosity, I'll update my Taranis Plus from OTX to Edge, in light of Groove70's comment, to see what happens, and respond back here.

Groove70, 2023-11-15:
What I can reproduce regardless of restarting/rebooting the device X7 with 2.9.2:
TRAINER/CPPM Power on the module pins (didn't work in 2.9.1).
TRAINER/SBUS No Power on module pins.
I always crosscheck with External Module ON where I have power on the pins (external receiver led).

PR #4326 (2023-11-15): fix: external module power for SBUS trainer
Please note that this PR is explicitely targeted at `2.9`, as the `main` should be fixed with the more generic PR (see #4266)

Fixes #4324
closed 2023-11-16 by raphaelcoeffic