EdgeTX/edgetx #4325: Horus X12S echo on sound system after loading nightly.

Kevltan, 2023-11-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

There is now an echo on the sound system of my X12S having loaded the nightly. It does not do this in 2.9.2.
 I loaded a nightly about 2 months ago and the issue was there then, but due to other issues and commitments I returned back to 2.9.0. I was hoping that someone else would spot this and it would be ok the next time I loaded a nightly, but it seems unresolved and has gone under the radar.

### Expected Behavior

An echo (Horus has stereo capabilities) so there maybe a delay on the stereo channels.

### Steps To Reproduce

An echo on all when a voice is called (Horus has stereo capabilities) so there maybe a delay on the stereo channels.

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X12

### Anything else?

latest nightly

pfeerick, 2023-11-17:
@raphaelcoeffic Maybe https://github.com/EdgeTX/edgetx/pull/3957 is responsible?

Kevltan, 2023-12-11:
Anybody had a chance to look at this problem please??

richardclli, 2023-12-13:
Let me do a test to see if I can repeat the problem, I got a X12S.

richardclli, 2023-12-13:
I got your point, it seems the sound effect is quite different between these 2 versions.  I checked the source code, the audio driver didn't changed between these 2 versions, so......

richardclli, 2023-12-13:
This is the 2.9 sound
https://github.com/EdgeTX/edgetx/assets/18414861/cef3fd61-2950-40bc-b1f2-427a47e34715

This is the 2.10 sound
https://github.com/EdgeTX/edgetx/assets/18414861/d269bb61-23d6-48e5-946b-51e3f86cd34e

richardclli, 2023-12-13:
I check the source code and the change is to merge the horus audio driver with nv14 audio driver and in which both uses the same audio chip.

Kevltan, 2023-12-13:
Sounds good, please let me know how you get on and if it resolves the issue.

Kevltan, 2023-12-18:
Has anybody had any luck yet with resolving this problem with the sound on the X12S ??

richardclli, 2023-12-20:
No luck, the new driver didn't changed much.  And both X12S and EL18 uses the same driver, but didn't spot any problem in EL18.

Kevltan, 2023-12-20:
Does the EL18 have stereo capabilities like the X12S?

Kevltan, 2023-12-20:
Looks like a mono speaker output on the back of the EL18. There are 2
independently driven stereo speakers on the X12S which is where I think the
problem is caused. I think there is posdibly a delay in sound between the 2
channels.

richardclli, 2023-12-21:
I do not have the schematic of X12S, it is hard to tell what would be the differences.  But the driver didn't change anything special, maybe I will see if I can revert the old driver for X12S build to see if there are any differences.  Do you know how to build a firmware from a branch?

pfeerick, 2023-12-21:
Don't know if this helps you at all @richardclli 
https://github.com/opentx/opentx/issues/8054#issuecomment-717622677

There is also some PCBX12S specific code in the vs1053b driver file. I would be curious to know if we can pin down #3957 as being responsible for this change... as then at least we'd know there to start.

raphaelcoeffic, 2023-12-21:
> This is the 2.9 sound https://github.com/EdgeTX/edgetx/assets/18414861/cef3fd61-2950-40bc-b1f2-427a47e34715
> 
> This is the 2.10 sound https://github.com/EdgeTX/edgetx/assets/18414861/d269bb61-23d6-48e5-946b-51e3f86cd34e

Am I the only one to not hear any difference between the two?

3djc, 2023-12-21:
> > This is the 2.9 sound https://github.com/EdgeTX/edgetx/assets/18414861/cef3fd61-2950-40bc-b1f2-427a47e34715
> > This is the 2.10 sound https://github.com/EdgeTX/edgetx/assets/18414861/d269bb61-23d6-48e5-946b-51e3f86cd34e
> 
> Am I the only one to not hear any difference between the two?

You are not the only one

Kevltan, 2023-12-21:
Only on the X12S does issue exist. Sounds like the radio has been put and a
large biscuit tin!!

richardclli, 2023-12-21:
> > This is the 2.9 sound https://github.com/EdgeTX/edgetx/assets/18414861/cef3fd61-2950-40bc-b1f2-427a47e34715
> > 
> > This is the 2.10 sound https://github.com/EdgeTX/edgetx/assets/18414861/d269bb61-23d6-48e5-946b-51e3f86cd34e
> 
> Am I the only one to not hear any difference between the two?

Yes, no differences in these 2 samples.  The "echo" effect only present in my first flash, but not recorded.  May try to test again this weekend.

richardclli, 2023-12-22:
I have made a branch with old audio driver, please see if this resolves the problem first so that we can see if we can figure out what cause the problem mentioned.

https://github.com/EdgeTX/edgetx/tree/x12s-old-audiodrv

richardclli, 2023-12-22:
I spot the softReset code is different for nv14 and x12s target and the new driver did not follow either nv14 nor x12s, not sure if this maybe the cause.

Kevltan, 2023-12-22:
There doesn't seem to be an option of a bin file download to test the firmware branch - or am I doing something wrong here?

pfeerick, 2023-12-22:
> There doesn't seem to be an option of a bin file download to test the firmware branch - or am I doing something wrong here?

Since it is a branch, rather than a PR, you  would either need to use gitpod to build it, or wait for someone to build it for you.

Kevltan, 2023-12-22:
How do I carry this branch across to the GitHub ready to code environment so that I can build this branch? Or do I have to carry across the edits/changes to the environment and then build?

pfeerick, 2023-12-23:
From the wiki instructions:
> To start building EdgeTX firmware for your radio, open the following link with the web browser of your choice:
> https://gitpod.io/#https://github.com/edgetx/edgetx/tree/main

notice the end of the url says `/tree/main`? Try changing that to `/tree/x12s-old-audiodrv` ;)

richardclli, 2023-12-23:
I think the driver in 2.9 and main intentionally init the audio chip with minimal spatial effect (SM_EARSPEAKER_LO), however maybe the old driver in 2.9 somewhat failed to init the audio chip in this mode and the new driver in main will ensure the init is done.

Kevltan, 2023-12-23:
Yep!! the above works just remove 
         | SM_EARSPEAKER_LO 
from the driver and build.....
It works all spatial effect now gone! 😊👍👍

richardclli, 2023-12-27:
Cahnge in line 54 is not necessary.

Kevltan, 2023-12-27:
> Change in line 54 is not necessary.

Just did it to reduce binary size a tiny bit if not used.

richardclli, 2023-12-30:
#define will not affect binary size anyway, the compiler just did string replacement in the source file before compilation.

Kevltan, 2023-12-30:
Ok thanks for letting me know that and thanks for all your help😊👍

raphaelcoeffic, 2023-12-31:
Ok, so bottom line is: we should remove `SM_EARSPEAKER_LO` from the init code? Assuming that it was actually never in effect with the X12 as the first init failed somehow and was never retried?

Kevltan, 2023-12-31:
Edit works fine with my X12S and resolves the "spatial reverb" effect and
makes the sounds "clean" again as in 2.9.2. So from me it is a YES.👍

PR #4492 (2023-12-31): fix(vs1053b): Remove EarSpeaker Spatial Processing
See section *9.3 EarSpeaker Spatial Processing* of the VS1053b datasheet.

Fixes #4325
closed 2023-12-31 by pfeerick