EdgeTX/edgetx #4338: Newly generated input not available as mixer input (2.9.2)

mha1, 2023-11-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Newly generated input not available as mixer input

### Expected Behavior

Newly generated input available as mixer input

### Steps To Reproduce

1. create new blank model
2. go to model settings / inputs tab
3. observe predefined thr, ail, elev and rud inputs
4. add input 06 (yes, leave out 05) and name it I6. Leave all as is.
5. goto mixer tab and add CH6 mixer
6. goto source, filter for inputs
7. observe input I6 missing
8. goto manage models and select any other model
9. select previous model again
10. goto model settings / mixer tab and repeat step 5. and 6.
11. observe I6 beging present for selection

### Version

2.9.1

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

2.9.2 (not option to select 2.9.2 in issues)

pfeerick, 2023-11-19:
Already fixed in main via #4317 and #4330, and cherrypicked in to 2.9 as b359be35460316ec0e32d227d92b2060e762b72c

PS: You *can* edit issues after making them, if you need to change something that the template won't let you select / enter.

closed 2023-11-19 by pfeerick

mha1, 2023-11-19:
great and sorry, couldn't find 4317