EdgeTX/edgetx #4339: Throttle warning still acts on throttle stick after throttle source assigned to CH5

mha1, 2023-11-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Consider the following minimal helicopter setup with 5 mixers defined:
- CH1: Input 1 - Thr stick used as collective pitch control
- CH2/3: Inputs 2 and 3 - Ail, Ele sticks , used for roll/pitch axis control
- CH4: Input 4 - Rud stick used as rudder control
- CH5: Input 5 -  switch SG to control governer (off, idle up 1, idle up2)

Throttle set to CH5 (the governer output). All preflight checks disabled but throttle state. 

Issue: the throttle warning when selecting the model still acts on the throttle stick. Please see attached minimal model.yml

[image]
[image]

### Expected Behavior

Throttle warning acts on CH5 state

### Steps To Reproduce

see problem description and 
[model33.yml.txt](https://github.com/EdgeTX/edgetx/files/13403017/model33.yml.txt)

### Version

2.9.1

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

2.9.2 (can't select 2.9.2 in issues)

Note: what does "0" mean?
[image]

raphaelcoeffic, 2023-11-19:
This is basically a *known issue*: the throttle warning code is pretty stupid and will only check in stick itself and ignore the throttle channel.

See here:
```c++
  // throttle channel is either the stick according stick mode (already handled in evalInputs)
  // or P1 to P3;
  // in case an output channel is choosen as throttle source (thrTraceSrc>NUM_POTS+NUM_SLIDERS) we assume the throttle stick is the input
  // no other information available at the moment, and good enough to my option (otherwise too much exceptions...)
  uint8_t thrchn = ((g_model.thrTraceSrc==0) || (g_model.thrTraceSrc>NUM_POTS+NUM_SLIDERS)) ? THR_STICK : g_model.thrTraceSrc+NUM_STICKS-1;

  if (!mixerTaskRunning()) getADC();
  evalInputs(e_perout_mode_notrainer); // let do evalInputs do the job

  int16_t v = calibratedAnalogs[thrchn];
  if (g_model.thrTraceSrc && g_model.throttleReversed) { // TODO : proper review of THR source definition and handling
    v = -v;
  }

  if (g_model.enableCustomThrottleWarning) {
    int16_t idleValue = (int32_t)RESX * (int32_t)g_model.customThrottleWarningPosition / (int32_t)100;
    return abs(v - idleValue) > THRCHK_DEADBAND;
  }
  else {
    return v > THRCHK_DEADBAND - RESX;
  }
```

mha1, 2023-11-19:
Ok, because the mixer task is not yet running?

raphaelcoeffic, 2023-11-19:
> Ok, because the mixer task is not yet running?

This. That would require triggering the full cycle instead of just evaluating the inputs.

mha1, 2023-11-19:
would be worth mentioning the rather unexpected behavior for selecting channels as throttle source: https://edgetx.gitbook.io/edgetx-user-manual/edgetx-user-manual/user-manual-for-color-screen-radios/model-settings/model-setup/throttle

Something like this:

`Source: The source that will be used for the throttle. In case an output channel is chosen as throttle source it is assumed the throttle stick is the input, i.e. the throttle stick will be evaluated.`

johnjoeallen, 2024-01-27:
I get the Throttle not idle 1%, the warning goes when the throttle stick is a 0. Surely the check should be for -100

johnjoeallen, 2024-01-27:
> I get the Throttle not idle 1%, the warning goes when the throttle stick is a 0 (centered). Surely the check should be for -100

Turn off custom position, that was the problem :)

mha1, 2024-04-25:
@pfeerick @philmoz 
Can we at least try to make this a little better, by checking the real throttle input? 
I believe by extending the source selection options to physical inputs, ie sticks, switches, 6Pos, pots and also removing the channel options we'd cover 99% of all reasonable use case.

arnaudmz, 2024-04-27:
Hi there. I just tested and confirm that using another input as throttle (S1, S2…) is correctly handled on OpenTX (tested on `2.3.15`) and generates a warning adequately at startup.

I'd love to see at least this behaviour.

mha1, 2024-10-13:
> @pfeerick @philmoz Can we at least try to make this a little better, by checking the real throttle input? I believe by extending the source selection options to physical inputs, ie sticks, switches, 6Pos, pots and also removing the channel options we'd cover 99% of all reasonable use case.

anything planned for this issue?