EdgeTX/edgetx #4340: More models on model select screen

Mike-HRC, 2023-11-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

This is particularly related to the Colour TX16S touchscreen.

Can we have more models shown on the Model Select screen please - a lot of us have come from OpenTx where we have 8 models shown on the model slect screen to chose from - with only 4 shown on the ETX screen it is a bit cumbersome to scroll through lots of pages of 4 to find the one that I want - it would be much better with 8 shown at a time (like OpenTx - which still has the name and picture for each).

I know that I could split the category into smaller groups - but this is still extra faff for th euser to set up and move models around - it would be much easier just to keep larger category groupings but just show more models at one.

I hope this makes sense, and it is something that you can - this is one of the few things that keeps me using OpenTx as my "daily driver" - thanks in advance.

### Describe the solution you'd like

Show 8 models on the screen at once (ideally with Name and picture), allow us to select any of them and scroll through a currently

raphaelcoeffic, 2023-11-20:
You should use labels to make better categories instead of scrolling through endless lists ;-) This is much better.

Mike-HRC, 2023-11-20:
The categories are all a bit clunky as well though - if I have 2 categories selected then nothing showa - still if I have say 12 models in a category then I have to scroll down through 3 pages.  If EdgeTx had 8 per page and I could scroll left for the next screen that woul dbe ideal.

Having more categories that already don't work well doesn't help anything :)

gagarinlg, 2023-11-20:
The categories are a filter. When you select two, only models that are on both categories, will be shown.

closed 2024-04-08 by pfeerick