EdgeTX/edgetx #4343: Insta trim shouldn't adjust your throttle

Mantamatt1, 2023-11-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When using insta trim it is also trimming the throttle.

### Expected Behavior

Insta trim should not trim the throttle as it has not done it until I updated firmware.  as I was about 80% throttle when I done Insta trim. I couldnt shut the motor down. I didn't realise what happened till the motor cut out and I could land! Not a nice experience lol

### Steps To Reproduce

Set up insta trim in special functions, move throttle stick up and use insta trim you will see throttle trim will set as well

### Version

2.9.2

### Transmitter

RadioMaster TX16S / TX16SMK2

pfeerick, 2023-11-20:
I can't repo this on v2.9.2 w/ TX16S... instant trim is correctly not altering the throttle trim for me. 

Are you sure this was v2.9.2, and not a nightly build for 2.10? What version does it say you're running in radio settings -> version (last tab)? As I do seem to be able to reproduce this there. 

If this was v2.9.2, can you upload a zip with a copy of your radio and models folder, as it may be caused by a specific configuration?

gagarinlg, 2023-11-20:
The configuration from the nightlies is not backwards compatible to 2.9.
You should always make a backup!

Mantamatt1, 2023-11-20:
That will explain it! I'll remember to do that in future.
This can be closed now.
Thank you

pfeerick, 2023-11-20:
> My bad. I was on a nightly! So sorry!
> I will mention though the flash for 2.9.2 stops your sticks pots and
> sliders.
> You have to goto... radio settings/hardware, so you can re enable pots and
> sliders and the calibration to get them working again.
> And any thing associated with these switches has to be re assigned. I.e if
> you have volume control on a pot, you will have to re assign that pot.
> Apologies if this has already been mentioned.
> Thanks for all your great work 👍

No problem, you still caught a 2.10 big for us :)

That is to be expected (loss of pot and slider config and analog
calibration)... You should be able to upgrade without any major settings
issues, but downgrading without using a backup of settings may (and
definitely will in the case of 2.10 -> 2.9) result in settings loss or
corruption. Software is generally backwards compatible, not forward
compatible.. ;)

closed 2024-04-08 by pfeerick