EdgeTX/edgetx #4346: The use of TEXT_SIZE in a Widget.

rdeanchurch, 2023-11-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

There are tow areas of concern.
1. The return from the option TEXT_SIZE is in the flags field for INVERS, and Justification,..., it needs to be shifted left 8 bits to be usable as a FontSize.
2.  I cannot discover a method of setting the default value to be displayed in the GUI. (of the range values)
Setting an integer is required, but what the integer seems to have no affect.
It is also interesting to note that the sizes of those fonts from TEXT_SIZE includes another font size than those 
indicated by font constants (XXS).  Using names different than font constants also seems inconsistent.

My discoveries:
DEFSIZE = 0             --        Reasonable y adj. for a text line
  --                       lcd.sizeText(h              ↓                         likely to support 
  --                wgt.option-              ↓       ↓      ↓  to show "RxBt  5.200V"  
   -- XXLSIZE  = XXLSIZE --  __      76     54     392, 54              x Col. by y rows         
   -- DBLSIZE  = DBLSIZE --    XL   39     28    201, 28               1           6
   -- MIDSIZE  = MIDSIZE --     L   29     21   145, 21                2           8
local BLDSIZE  = BOLD    -- BOLD  20     15    96, 15                3          12
local DEFSIZE  = 0       --        STD   21    16    95, 16                3          11
   -- SMLSIZE  = SMLSIZE --       Xs 17     12    74, 12                4         14  
local XSMSIZE  = 512     --      XXS  12      8    55, 12                 5         18

### Expected Behavior

1 I would expect a FONT-SIZE constant  for a return value.  Ready to be used in lcd.drawText(
2 Maybe just documentation to explain how to set default, Minimum and Maximum is all that is needed.
  I'm unable to find the code, if I did I probably wouldn't be able to decipher it.

### Steps To Reproduce

lua Widget with 
local options = {
...
  { "TextSize",  TEXT_SIZE, 3,1,7 },  -- #### Remove after auto text size is implemented ####

### Version

2.9.2

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

v2.9.2

### Anything else?

Limited testing on the tx shows consistent behavior with EdgeTxCompanion.  
Both on v2.9.2

My ability to do a search seems completely inadequate.

There may well be an Issue that I did not find.

pfeerick, 2023-12-21:
Have you by any chance looked at this again with a recent nightly for 2.10? As #4394 fixed an issue around the TextSize option type that may be relevant here.

rdeanchurch, 2023-12-21:
I just looked at it Companion 2.10.0 and do not see a change in either setting the Default value or in the return value being
in the wrong bit position.  I made the assumption that the min and max value would be 0 to 6 or 1 to 7 as there are seven
possible selections for textsize.

I do not have a transmitter running 2.10 at this time.

philmoz, 2024-08-04:
Changing the return value from TEXT_SIZE would break existing scripts.
I would suggest adding a new option type (e.g. FONT), that returns the value shifted for use as a font.

The third parameter of the option will set the default font size selected in the popup menu (tested in 2.9.2 and later).

You cannot set the min and max values for the TEXT_SIZE option - it will always allow the user to choose from all available fonts.