EdgeTX/edgetx #4348: FRSKY RX6R not able to flash OTA

craigywsm, 2023-11-21:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

With Reference to https://github.com/EdgeTX/edgetx/pull/4077

i can confirm that when using Archer or Archer Plus receivers it does indeed include the option to flash OTA

However for the RX series of receiver e.g. RX6R there is no option to flash OTA as they have the option off accst or access firmware.

### Describe the solution you'd like

For RX series to have the OTA option (RX6R etc)

pfeerick, 2023-11-22:
Does FrSky actually say this receiver is OTA update-capable? I'm not seeing
any mention of that on the product specs, or the OTA logo that they usually
show on the casing. Perhaps this is a SPort-only updated RX?

craigywsm, 2023-11-22:
@pfeerick i will double check when i get home but I am sure Ethos allows the update via OTA.
I am sure its all access receivers e.g. baked into access as I have an old R9M and it allows OTA

ParkerEde, 2023-11-22:
I am very sure that the "old" RX6R and RX4R do not support OTA. This was first introduced with the Archer receivers.

pfeerick, 2023-11-22:
> I am sure its all access receivers e.g. baked into access as I have an
old R9M and it allows OTA

I thought so too at first, but if you read the ACCESS FAQs on the FrSky
website it actually points out that not all ACCESS features are available
for all receivers (and transmitters)... and there is a subtle reference to
OTA there. R9 is different again, and the receivers either have OTA in the
name, or clearly show/list that capability. But, yes, please do confirm
against Ethos (and that it actually works if you are able).

> The OTA function is also one of ACCESS functions, and it mainly relies on
the memory resource of the receiver, this function cannot be enabled if the
receiver memory is insufficient.

craigywsm, 2023-11-23:
@ParkerEde @pfeerick 
Some interesting information in the way the latest version of Ethos 1.4.14 behaves
an old RX8RPRO (accst RX) no option for OTA
[image]

R9MM /mini (old non OTA version) no option for OTA
[image]

RX6R (access firmware) has option for OTA
[image]

Here is where it gets interesting RX6R (accst firmware) Has the OTA option ???
[image]

It looks like to me it is looking at the RX of the firmware and saying does it support access of so allow OTA

It looks like https://github.com/EdgeTX/edgetx/pull/4077 is slightly off

pfeerick, 2023-11-23:
But does the option actually work? ... i.e. the option may be shown, but does it actually work, or is that a bug in ETHOS?

It probably uses the same checks we do (which was changed in #4185), and is simply one of "what product ID is this?", and "are we running a supported OTA RF module?" . There is zero communication with the RX, and no consideration of what firmware it is currently running... which means we'll probably end up showing the option without it actually being able to work also at some point. 

btw, what radio is this with (the one not showing the OTA option), as you never mentioned?

https://github.com/EdgeTX/edgetx/blob/b359be35460316ec0e32d227d92b2060e762b72c/radio/src/io/frsky_firmware_update.h#L80-L87

craigywsm, 2023-11-24:
@pfeerick from memory it does work as I updated all my access RX6R's let me double check this though.
Sorry the radio is X9D2019

craigywsm, 2023-11-25:
> But does the option actually work? ... i.e. the option may be shown, but does it actually work, or is that a bug in ETHOS?
>
> It probably uses the same checks we do (which was changed in #4185), and [...]

@pfeerick after some testing on ethos your right it does not flash OTA, it recognises the rx but the receiver never accepts the firmware. 
I do apologise for waisting your time. And thank you for the time you have put into this ticket

ParkerEde, 2023-11-29:
@craigywsm so could this ticket be closed?

craigywsm, 2023-11-29:
Yes

closed 2023-11-29 by 3djc