EdgeTX/edgetx #4350: No GPS Telemetry on the Jumper T20S

Dist0rtion, 2023-11-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I think there is a bug/error with the Jumper T20S and the current nightly build.
I do not get any GPS telemetry displayed.
I have tested the current nightly build on the Jumper T-Pro and the Radiomaster TX16S, both show full telemetry including GPS.
_(Could there be an error with ExpressLRS? Because different receivers.)_

The OSD of Betaflight still shows me GPS data.
LAT/LONG/SATS/kmh and so on.
But nothing on the radio

### Expected Behavior

Get GPS Telemetry on the Radio.

### Steps To Reproduce

-

### Version

Nightly (Please give date/commit below)

### Transmitter

Other (Please specify below)

### Operating System (OS)

Windows

Dist0rtion, 2023-11-21:
Fixed it by my own.. 
Build it with Gitpod with following configuration.
`cmake -Wno-dev -DPCB=X7 -DPCBREV=T20 -DDEFAULT_MODE=2 -DPPM_UNIT=METRIC DLUA=YES -DLUA_MIXER=YES -DGVARS=YES -DHELI=NO -DJUMPER_RELEASE=YES -DCMAKE_BUILD_TYPE=Release ../`

3djc, 2023-11-30:
There is no reason for t20 to behave differently that any other 128x64 radio in that area, since telemetry code is 100% the same

Dist0rtion, 2023-12-01:
Found out that I might have a defective T20S.
Probably the internal ELRS module.
Can't explain why a large part of the telemetry is transmitted and displayed, except GPS.

I have tested a new T20S and it works.
_(Both T20S remote controls run exactly the same firmwares.)_

Can be closed.

closed 2023-12-01 by 3djc