EdgeTX/edgetx #4352: Companion Splash screen editor not able to apply image to firmware for BW Radio X9D 2019

craigywsm, 2023-11-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

When using companion 2.9.2 (not work on others either)
I get and error "Compress image size exceeds reserved space" when trying to save the firmware when copying the image over and then saving the firmware.
[image]

### Expected Behavior

With the same image in other systems this works.

### Steps To Reproduce

If I use the Settings--> Edit radio Splash Screen
Left section I load the firmware from radio (with Edge splash screen)
Right Section I load the picture with 212x64 (8 bit) 5KB
click the arrow to copy the image from right to left
Then Click save on the left section
[image]

Here is the same thing if using OTX
[image]

Also when picking the image when writing the flash to the radio it also does not work. 
[image]
it still uses the edge tx logo

### Version

2.9.2

### Transmitter

FrSky X9D+2019

### Operating System (OS)

Windows

### OS Version

Windows 10 x64

pfeerick, 2023-11-22:
This is most likely an issue that can only be resolved in the short term by
reducing the size/complexity of your splash screen so that it does fit. The
reserved space for splash screens is slightly smaller (X9D family radios
were running out of flash memory space, so it was either RLE compression or
no custom splash screen at all). If selecting a too-large image when
flashing the firmware fails silently and uses the stock image, rather than
halting with the same error, that is a bug in Companion though IMO.

craigywsm, 2023-11-22:
@pfeerick Thanks for your input.
Yes it does look like if I reduce it down to 1BIT PNG then it doe save.
Here is the image for reference:
[image]
It would be interesting to know the file size of the  stock edge TX splash logo.
Edge are no longer saving the models to the flash memory, they are now in yaml in the sd card should this not provide more storage?
interested to know your thoughts

gagarinlg, 2023-11-22:
Multiple answers:
For the image size, there is no real answer. The images are bein compressed and the resulting size depends on the image contents

The models have never been stored in the same memory as the firmware. The models were stored in an EEPROM that is an extra IC in the radio and thee firmware is stored in the flash memory in the processor

craigywsm, 2023-11-22:
thanks @gagarinlg for that information.
I see what you mean there is no real answer for this one other than reducing the size of the image

craigywsm, 2023-11-25:
After some testing I can get an image to work but it needs to be under 1KB 
Is there anything that can be done to allow more space for the image?

gagarinlg, 2023-11-25:
Only be reducing the features for B&W radios